platformio/platformio-core #1993: Home: Could not load boards list

mbariola, 2018-12-27:
Server error: {"message":"No module named _dumbwin32proc","args":["No module named _dumbwin32proc"],"type":"ImportError"}

ivankravets, 2018-12-27:
What is your Windows version? Is it 32bit or 64bit?

mbariola, 2018-12-27:
64bit on a 64bit vscode

ivankravets, 2018-12-27:
Please type in PlatformIO IDE Terminal 
```
pio --version
```

Which version do you use?

ivankravets, 2018-12-27:
Also, could you try to remove `%USER_HOME%/.platformio/contrib-pysite` folder and restart VSCode?

mbariola, 2018-12-27:
PlatformIO, version 3.6.4b1

unrelated but maybe of interest to you, I mistakenly wrote pio --versionù and got an error that might need a more informative handling (e.g. unrecognized option maybe?)
` pio --versionù
Error: Traceback (most recent call last):
  File "c:\users\massi\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\massi\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\massi\.platformio\penv\lib\site-packages\click\core.py", line 679, in main
    with self.make_context(prog_name, args, **extra) as ctx:
  File "c:\users\massi\.platformio\penv\lib\site-packages\click\core.py", line 594, in make_context
    self.parse_args(ctx, args)
  File "c:\users\massi\.platformio\penv\lib\site-packages\click\core.py", line 989, in parse_args
    return Command.parse_args(self, ctx, args)
  File "c:\users\massi\.platformio\penv\lib\site-packages\click\core.py", line 854, in parse_args
    opts, args, param_order = parser.parse_args(args=args)
  File "c:\users\massi\.platformio\penv\lib\site-packages\click\parser.py", line 208, in parse_args
    self._process_args_for_options(state)
  File "c:\users\massi\.platformio\penv\lib\site-packages\click\parser.py", line 234, in _process_args_for_options
    self._process_opts(arg, state)
  File "c:\users\massi\.platformio\penv\lib\site-packages\click\parser.py", line 355, in _process_opts
    self._match_long_opt(norm_long_opt, explicit_value, state)
  File "c:\users\massi\.platformio\penv\lib\site-packages\click\parser.py", line 265, in _match_long_opt
    raise NoSuchOption(opt, possibilities=possibilities)
  File "c:\users\massi\.platformio\penv\lib\site-packages\click\exceptions.py", line 145, in __init__
    UsageError.__init__(self, message, ctx)
  File "c:\users\massi\.platformio\penv\lib\site-packages\click\exceptions.py", line 38, in __init__
    ClickException.__init__(self, message)
  File "c:\users\massi\.platformio\penv\lib\site-packages\click\exceptions.py", line 14, in __init__
    message = message.encode('utf-8')
UnicodeDecodeError: 'ascii' codec can't decode byte 0xf9 in position 25: ordinal not in range(128)

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
`
I have no such folder in my `%USER_HOME%/.platformio` path.. I have such a folder in `%USER_HOME%/.platformio/packages/` . Shall I delete that one?

ivankravets, 2018-12-27:
Yes, please remove `%USER_HOME%/.platformio/packages/contrib-pysite` and restart VSCode.