platformio/platformio-core #2017: Home: Could not load mDNS services

mecki1101, 2019-01-03:
PIO Core Call Error: "<Exception instance at 0x5f45748 with str error:\n Traceback (most recent call last):\n  File \"C:\\Users\\mecki\\.platformio\\packages\\contrib-pysite\\twisted\\python\\reflect.py\", line 448, in safe_str\n    return str(o)\nUnicodeEncodeError: 'ascii' codec can't encode character u'\\xfc' in position 1780: ordinal not in range(128)\n>"

ivankravets, 2019-01-09:
Could you open PlatformIO IDE Terminal and type
```
pio device list --mdns --json-output
```

Please share an output here.

OttoWinter, 2019-02-21:
I recently played around with mDNS and zeroconf myself. AFAIK this happens when your computer has multiple IP addresses on the same network interface and one of them is invalid. Can you go into your windows network settings (under advanced TCP/IP settings) and check if you have multiple IP addresses assigned to the same interface?

In my own library I ended up just monkey-patching that method so that errno 10022 would get ignored (like here https://github.com/jstasiak/python-zeroconf/blob/master/zeroconf.py#L1756) - the other IP address can still listen for multicast traffic on the subnet.

ivankravets, 2019-05-03:
@mecki1101 Have you resolved this issue?

mecki1101, 2019-05-04:
Hi,yes the problem is solved. Thx for your help and sorry for the late answer.Kind regards

closed 2019-05-04 by ivankravets