platformio/platformio-core #2031: OS_REVEAL_FILE Error, dev core 4.0.0a1, Python3, VSCode IDE

meltdown03, 2019-01-12:
### Configuration

**Arch Linux x86_64**:

**Git version 4.0.0a1** All using **Python 3, not Built-In Core**, VSCode):
Installed via `git clone https://github.com/platformio/platformio-core.git && python3 -m pip install --user -e ./platformio-core/`

### Description of problem
In PIO Home -> Libraries -> Installed Libraries, clicking "Reveal" brings up error in VSCode:

OS_REVEAL_FILE
Invalid params: {"type":"TypeError","args":
["startswith first arg must be bytes or a tuple
of bytes, not str"], "message":"startswith first
arg must be bytes or tuple of bytes, not str"}

I believe it's another Python 3 related issue. When using the _Built-In Development Core_ and the default python2 included in the .platformio/penv, it doesn't happen.

meltdown03, 2019-01-13:
Here is a more detailed error from running `pio home` via the terminal:

API Exception: {'type': 'TypeError', 'args': ('startswith first arg must be bytes or a tuple of bytes, not str',), 'message': 'startswith first arg must be bytes or a tuple of bytes, not str'}
Traceback (most recent call last):
  File "/home/neal/.platformio/packages/contrib-pysite/jsonrpc/manager.py", line 112, in _get_responses                        
    result = method(*request.args, **request.kwargs)
  File "<string>", line 254, in reveal_file
  File "/home/neal/.local/lib/python3.7/site-packages/click/termui.py", line 480, in launch                                     
    return open_url(url, wait=wait, locate=locate)
  File "/home/neal/.local/lib/python3.7/site-packages/click/_termui_impl.py", line 465, in open_url                            
    url = os.path.dirname(_unquote_file(url)) or '.'
  File "/home/neal/.local/lib/python3.7/site-packages/click/_termui_impl.py", line 437, in _unquote_file                       
    if url.startswith('file://'):
TypeError: startswith first arg must be bytes or a tuple of bytes, not str

ivankravets, 2019-04-18:
Thanks for the report!
Fixed in the latest PIO Plus Core. Please re-run `pio update`.

closed 2019-04-18 by ivankravets