platformio/platformio-core #2032: Home: Could not load platform data

anysomewhere, 2019-01-12:
PIO Core Call Error: "Platform C:\\Users\\hey\\.platformio\\platforms\\espressif32\r\n--------\r\nUpdating espressif32                     @ 1.5.0          [1.6.0]\r\r\nUninstalling espressif32 @ 1.5.0: \t[OK]\r\nPlatformManager: Installing espressif32 @ 1.6.0\r\r\nDownloading\r\nDownloading\r\nWarning! Package Mirror: The size (76725 bytes) of downloaded file 'platform-espressif32-1.6.0.tar.gz' is not equal to remote size (75277 bytes)\r\nLooking for another mirror...\n\nError: Could not install 'espressif32' with version requirements '1.6.0' for your system 'windows_amd64'.\r\n\r\nPlease try this solution -> http://bit.ly/faq-package-manager"

chr2002qw, 2019-01-14:
Same Issue here

ivankravets, 2019-01-15:
Should be fixed. Sorry for the issue.

closed 2019-01-15 by ivankravets