platformio/platformio-core #2038: Home: Could not initialize project

Moerup, 2019-01-14:
Trying to create a new mbed project on the F746G Discovery board and getting the following error every time.

Tried latest release 3.6.4 and the dev version 4.0.0a1

`PIO Core Call Error: "<Exception instance at 0x4b36288 with str error:\n Traceback (most recent call last):\n  File \"C:\\.platformio\\packages\\contrib-pysite\\twisted\\python\\reflect.py\", line 448, in safe_str\n    return str(o)\nUnicodeEncodeError: 'ascii' codec can't encode character u'\\xf8' in position 58: ordinal not in range(128)\n>"`

Moerup, 2019-01-22:
If anyone else is having the same problem, I fixed it with creating a different folder for my project.
It was bugging out because it was in C:\Users\Mørup.
Didn't like the  "Ø"

ivankravets, 2019-01-23:
Duplicate of https://github.com/platformio/platformio-core/issues/143

closed 2019-01-23 by ivankravets