platformio/platformio-core #2063: Default linux proxy environment variables ignored

elestedt, 2019-01-23:
### Configuration

**Operating system**: Ubuntu 18.04 LTS

**PlatformIO Version** (`platformio --version`): 3.6.3

### Description of problem

When on linux behind a proxy, the OS default variables for setting a proxy (`http_proxy` and `https_proxy`) are ignored.
Instead it's required to manually set the upper case (Windows) variants. For a CLI this is trivial, but when the desktop environment launches apps such variables might not be possible for users to set. Especially not if IT controls all that the user does not have permissions to add session wide variables that need to be dynamic (because proxies might be selected on location on login).

#### Steps to Reproduce

1. Install PIO behind a proxy and try to run `pio platform install` or any other command that downloads data. I did this using the VSCode IDE.

### Actual Results

```
$ pio platform update
Platform Atmel AVR
--------
Updating atmelavr                        @ 1.12.1         [Off-line]
Updating toolchain-atmelavr              @ 1.50400.0      ^C^C
^C[Off-line]
Updating framework-arduinoavr            @ 1.10623.1      [Off-line]
Updating tool-avrdude                    @ 1.60300.2      [Off-line]

Failed to check for PlatformIO upgrades. Please check your Internet connection.
```

### Expected Results

```
$ pio platform update
Platform Atmel AVR
--------
Updating atmelavr                        @ 1.12.1         [1.12.2]
Uninstalling atmelavr @ 1.12.1:         [OK]
PlatformManager: Installing atmelavr @ 1.12.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
atmelavr @ 1.12.2 has been successfully installed!
Updating toolchain-atmelavr              @ 1.50400.0      [Up-to-date]
Updating tool-avrdude                    @ 1.60300.2      [Up-to-date]
Uninstalling framework-arduinoavr @ 1.10623.1:  [OK]
PackageManager: Installing framework-arduinoavr @ ~1.10623.2
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
framework-arduinoavr @ 1.10623.2 has been successfully installed!
```

Actual command to get this result was `HTTP_PROXY="<proxy URL>" pio platform update`

ivankravets, 2019-01-23:
I'm not sure that this is windows related environment variable. See http://docs.python-requests.org/en/master/user/advanced/#proxies

closed 2019-01-23 by ivankravets

elestedt, 2019-01-23:
So it's a python bug then?

ivankravets, 2019-01-23:
Please move this issue to https://github.com/requests/requests/issues

Actually, we use this package for any HTTP/S operations.