platformio/platformio-core #2066: Home: Libraries: Could not install library

brendandower90, 2019-01-23:
PIO Core Call Error: "Library Storage: C:\\Users\\Brendan\\AppData\\Roaming\\SPB_Data\\.platformio\\lib\r\nLibraryManager: Installing id=873\r\nDownloading...\n\nError: Traceback (most recent call last):\r\n  File \"c:\\users\\brendan\\appdata\\roaming\\spb_data\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 120, in main\r\n    cli(None, None, None)\r\n  File \"c:\\users\\brendan\\appdata\\roaming\\spb_data\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 700, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"c:\\users\\brendan\\appdata\\roaming\\spb_data\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 680, in main\r\n    rv = self.invoke(ctx)\r\n  File \"c:\\users\\brendan\\appdata\\roaming\\spb_data\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1027, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\brendan\\appdata\\roaming\\spb_data\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1027, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\brendan\\appdata\\roaming\\spb_data\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 873, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"c:\\users\\brendan\\appdata\\roaming\\spb_data\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 508, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"c:\\users\\brendan\\appdata\\roaming\\spb_data\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 26, in new_func\r\n    return f(get_current_context().obj, *args, **kwargs)\r\n  File \"c:\\users\\brendan\\appdata\\roaming\\spb_data\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib.py\", line 103, in lib_install\r\n    library, silent=silent, interactive=interactive, force=force)\r\n  File \"c:\\users\\brendan\\appdata\\roaming\\spb_data\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\lib.py\", line 352, in install\r\n    force=force)\r\n  File \"c:\\users\\brendan\\appdata\\roaming\\spb_data\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 723, in install\r\n    pkg_dir = self._install_from_piorepo(name, requirements)\r\n  File \"c:\\users\\brendan\\appdata\\roaming\\spb_data\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\lib.py\", line 213, in _install_from_piorepo\r\n    if app.get_setting(\"enable_ssl\") else dl_data['url'], requirements)\r\n  File \"c:\\users\\brendan\\appdata\\roaming\\spb_data\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 508, in _install_from_url\r\n    dlpath = self.download(url, tmp_dir, sha1)\r\n  File \"c:\\users\\brendan\\appdata\\roaming\\spb_data\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 211, in download\r\n    shutil.copy(dst_path, cc.get_cache_path(cache_key_data))\r\n  File \"C:\\Python\\Python27\\Lib\\shutil.py\", line 133, in copy\r\n    copyfile(src, dst)\r\n  File \"C:\\Python\\Python27\\Lib\\shutil.py\", line 97, in copyfile\r\n    with open(dst, 'wb') as fdst:\r\nIOError: [Errno 2] No such file or directory: 'C:\\\\Users\\\\Brendan\\\\AppData\\\\Roaming\\\\SPB_Data\\\\.platformio\\\\.cache\\\\e8\\\\fbe847ee5cca11466b644d2f689d15e8'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

brendandower90, 2019-01-24:
It seems that since the update I can't install any libraries. Perhaps my version of Platformio has been corrupted. I am running the library's source code locally instead now to circumvent this issue.

brendandower90, 2019-01-24:
Just some more info for you guys, I did a reinstall of platformio and this hasn't fixed the problem either.

ivankravets, 2019-01-29:
Could you try to remove `C:\\Users\\Brendan\\AppData\\Roaming\\SPB_Data\\.platformio\\.cache` folder and try again? It seems that your system antivirus blocks PlatformIO.

closed 2019-02-23 by ivankravets