platformio/platformio-core #2067: Home: Could not initialize project

ptracton, 2019-01-24:
Server error: {"message":"[Error 2] The system cannot find the file specified: u'C:\\\\Users\\\\ptrac\\\\Documents\\\\PlatformIO'","args":[2,"The system cannot find the file specified"],"type":"WindowsError"}

Taking a look at platformio for some embedded work, but it appears to fail to create projects.  Using the Atom IDE.  Home 2.0.0 Core 3.6.3.  Windows 10.  Examples do not open either.  Seems like this tool did not completely install or installed incorrectly.

ivankravets, 2019-01-29:
Could you reproduce this issue with PlatformIO IDE for VSCode?
- https://platformio.org/platformio-ide

ptracton, 2019-01-30:
Same issue with VSCode as with Atom.

[image]

ivankravets, 2019-01-30:
What is your %USERHOME%  directory? This wizard allows creating a project in another location. Please uncheck "Use default location". Do you see this error again?

ptracton, 2019-02-03:
%USERHOME% is not set.  The environment variable %USERPROFILE% has my home directory.   Is  this a windows version difference?

When I create a project and do not use the default location it at least makes a project, but there are no files in it.  I am attempting to make an STM32 project with the SPL (low level) device drivers.

ivankravets, 2019-02-23:
Did you resolve this issue?

ivankravets, 2019-03-07:
See https://github.com/platformio/platformio-core/issues/2195

The issue is linked with antivirus tools

closed 2019-03-07 by ivankravets