platformio/platformio-core #2070: `lib_archive = false` not disabling library archiving

jeffstaley, 2019-01-24:
Linker is finding `weak` symbols instead of the expected normal functions because framework libraries are archived. Placing `lib_archive = false` in my `platformio.ini` does not disable library archiving.

PlatformIO Version:  3.6.4

`platform.ini`:

```
[env:foo]
lib_archive = false
platform = ststm32
board = disco_f746ng
framework = arduino
build_unflags = -Os
build_flags = -g3 -Og -Wl,-Map,$BUILD_DIR/firmware.map  
```

jeffstaley, 2019-01-28:
You can test that `lib_archive = false` isn't working by using the above platform.ini and a dummy arduino program.

```
#include <ARDUINO.h>

void setup() {
}

void loop() {
}
```
Note that the linker is still being given the two libraries (`libFrameworkArduinoVariant.a` and `libFrameworkArduino.a`) as archives, *not as individual object files*. Which I presume is what the `lib_archive = false` option should do.

Here's my linker output:
```
arm-none-eabi-g++ -o .pioenvs/foo/firmware.elf -T ldscript.ld -Wl,-Map,/Users/jeffy/Documents/PlatformIO/Projects/test/.pioenvs/foo/firmware.map -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -mcpu=cortex-m7 --specs=nano.specs -Wl,--gc-sections,--relax -Wl,--check-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align .pioenvs/foo/src/test.cpp.o -L/Users/jeffy/.platformio/platforms/ststm32/ldscripts -L.pioenvs/foo -L/Users/jeffy/.platformio/packages/framework-arduinoststm32/STM32/variants/DISCO_F746NG -L/Users/jeffy/.platformio/packages/framework-arduinoststm32/STM32/CMSIS/CMSIS/Lib/GCC -Wl,--start-group .pioenvs/foo/libFrameworkArduinoVariant.a .pioenvs/foo/libFrameworkArduino.a -larm_cortexM7l_math -lc -lm -lgcc -lstdc++ -lc -Wl,--end-group
```

As to my more specific problem, I'm finding that when linking a dummy program, such as above, the linker is more often than not is finding the weak `SysTick_Handler` instead of the real one in `clock.c`.

As an alternatively to the `lib_archive = false` option, I can fix this issue by manually running the linker with the pair of linker options `--whole-archive` and `--no-whole-archive` surrounding the two library archives. This forces the linker to search the whole archive for the symbols and works fine. I just can't figure out how to do it automatically. I even tried using the advanced python scripting to do it automatically but I can't quite figure how to get only those two libraries and not libraries like `-lm`.  If you  include -lm and other libraries in that option pair, bad things happen.

Here a *manual* working linker call using the --whole-archive option
```
arm-none-eabi-g++ -o .pioenvs/foo/firmware.elf -T ldscript.ld -Wl,-Map,/Users/jeffy/Documents/PlatformIO/Projects/test/.pioenvs/foo/firmware.map -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -mcpu=cortex-m7 --specs=nano.specs -Wl,--gc-sections,--relax -Wl,--check-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align .pioenvs/foo/src/test.cpp.o -L/Users/jeffy/.platformio/platforms/ststm32/ldscripts -L.pioenvs/foo -L/Users/jeffy/.platformio/packages/framework-arduinoststm32/STM32/variants/DISCO_F746NG -L/Users/jeffy/.platformio/packages/framework-arduinoststm32/STM32/CMSIS/CMSIS/Lib/GCC -Wl,--start-group -Wl,--whole-archive .pioenvs/foo/libFrameworkArduinoVariant.a .pioenvs/foo/libFrameworkArduino.a -Wl,--no-whole-archive -larm_cortexM7l_math -lc -lm -lgcc -lstdc++ -lc -Wl,--end-group

```

ivankravets, 2019-02-23:
> But shouldn't the `lib_archive = false` option be making this unnecessary?

Good question... `lib_archive` option was intended for project library dependencies. Framework is not a library in the context of PIO Build System.

Parsing _LIBFLAGS is normal solution. We do similar things when need to do some patching of build process.

Could you share your final code of extra script? Maybe, this code will help other people.

Thanks!

jeffstaley, 2019-02-23:
> Good question... `lib_archive` option was intended for project library dependencies. Framework is not a library in the context of PIO Build System.

Then you might want to update the documentation since it sezs:

>Setting lib_archive = false will instruct PIO Build System to link object files directly (in-line). This could be useful if you need to override weak symbols defined in framework or other libraries.

Note the reference to framework in the documentation.

I'll post my workaround in a bit.

jeffstaley, 2019-02-23:
Here's the quick and dirty hack I used to overcome the issues of `weak` symbols not being found in the framework. It uses the advanced scripting feature of PlatformIO to wrap a linker `--whole-archive/--no-whole-archive` pair around the library archives.  The parsing isn't particular sophisticated and might be fooled in some cases, but it works for me so far.

extra_script.py:
```
from os import makedirs
from os.path import isdir, join
Import('env')

# bugfix for Weak symbols (https://github.com/platformio/platformio-core/issues/2070)
# Surround the library archives with linker --whole-archive/--no-whole-archive pair
libString = env.subst("$_LIBFLAGS") # something like "-Wl,--start-group .pioenvs/foo/libFrameworkArduinoVariant.a .pioenvs/foo/libFrameworkArduino.a -larm_cortexM7l_math -lc -lm -lgcc -lstdc++ -lc -Wl,--end-group"
tokens = libString.lstrip(' ').rstrip(' ').split()
archives = [t for t in tokens if t.endswith(".a")]		# find the archives
tokens = [t for t in tokens if t not in archives]			# remove the archives from the orignal set
#rebuild a new string with --whole-archive option pair surrounding the archives
newLibString = ' '.join([tokens[0] , "-Wl,--whole-archive"] + archives + ["-Wl,--no-whole-archive"] +tokens[1:])
env.Replace(
  _LIBFLAGS=newLibString,
  LINKCOM='$LINK -o $TARGET $LINKFLAGS $__RPATH $SOURCES $_LIBDIRFLAGS $_LIBFLAGS'
)
```