platformio/platformio-core #2077: К1986ВЕ92VE9x (Milandr) support and settings

nex0ma, 2019-01-27:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [x ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

ivankravets, 2019-01-29:
Could you provide more details?

ivankravets, 2019-02-02:
> RAK811

See https://github.com/platformio/platform-ststm32/issues/86. It was already added to STM32Duino and will be available in PlatformIO after the final releases. Stay tuned with us on https://twitter.com/PlatformIO_Org

> К1986

What is К1986? Do we have toolchains for it?

nex0ma, 2019-02-02:
Извините, пожалуйста, Иван, действительно, что такое K1986?
 
как я понимаю, это лицензированная серия клонов ARM Cortex (М0, M3,M4).Доступное здесь - https://www.milandr.ru/products/
либо здесь - https://ic.milandr.ru/products/mikrokontrollery_i_protsessory/32_razryadnye_mikrokontrollery/

Платы отладочные  у них дороги пока, поэтому может лучше просто сориентироваться на некоем generic  k1896. то есть как  у китайцев, покупаем контроллер, припаиваем к универсальной плате  для любых LQFP и начинаем прошивать через UART.  JTAG тоже представлен. Документация  у них, надеюсь, серьёзная и позволит создать конфигурацию для Миландровских контроллеров.

С большой надеждой к вам.

QuantScAv, 2021-12-26:
Hello, if the topic is relevant, then I can give an assembly for **K1986BE92QI**
 I would love to see it in the list of available platforms. Thank you

QuantScAv, 2021-12-26:
> > RAK811
> 
> 
> 
> See https://github.com/platformio/platform-ststm32/issues/86. It was already added to STM32Duino and will be available in PlatformIO after the final releases. Stay tuned with us on https://twitter.com/PlatformIO_Org
> 
> 
> 
> > К1986
> 
> 
> 
> What is К1986? Do we have toolchains for it?
> 
> 
> 
> 

Hello, if the topic is relevant, then I can give an assembly for **K1986BE92QI** (Cortex-M3)
 I would love to see it in the list of available platforms. Thank you

hww, 2023-02-22:
Is there a way to develop with platform for the K1986BE92QI nowadays?

EvgenyZn, 2023-10-25:
Здравствуйте! На 1986ВЕ4 ставлю опорный источник 1.23В, но он почему то не включается. Что делаю не так... На 3.3В всё норм работает...
// настройки по умолчанию для всех ADC
    ADC_StructInit(&ADC_InitStructure);
    ADC_InitStructure.ADC_StartDelay            = 0;
    ADC_InitStructure.ADC_TempSensor            = ADC_TEMP_SENSOR_Enable;               //ADC1_CFG_TS_EN
    ADC_InitStructure.ADC_TempSensorAmplifier   = ADC_TEMP_SENSOR_AMPLIFIER_Disable;     //ADC1_CFG_TS_BUF_EN
    ADC_InitStructure.ADC_TempSensorConversion  = ADC_TEMP_SENSOR_CONVERSION_Disable;    //ADC1_CFG_SEL_TS
    ADC_InitStructure.ADC_IntVRefConversion     = ADC_VREF_CONVERSION_Disable;           //ADC1_CFG_SEL_VREF
    ADC_InitStructure.ADC_IntVRefTrimming       = 0;
    ADC_InitStructure.ADC_IntVRefAmplifier      = ADC_INT_VREF_AMPLIFIER_Disable;        //ADC1_TRIM_SEL_VREF_BUF
    ADC_Init(&ADC_InitStructure);

    // настройка для ADC1
    ADCx_StructInit(&ADC1_InitStructure);
    ADC1_InitStructure.ADC_ClockSource       = ADC_CLOCK_SOURCE_CPU;
    ADC1_InitStructure.ADC_SamplingMode      = ADC_SAMPLING_MODE_SINGLE_CONV;
    ADC1_InitStructure.ADC_ChannelSwitching  = ADC_CH_SWITCHING_Disable;
    ADC1_InitStructure.ADC_Channels          = ADC_CH_ADC3_MSK | ADC_CH_ADC0_MSK | ADC_CH_ADC1_MSK;  // | ADC_CH_INT_VREF_MSK;
    ADC1_InitStructure.ADC_LevelControl      = ADC_LEVEL_CONTROL_Disable;
    ADC1_InitStructure.ADC_LowLevel          = 0;
    ADC1_InitStructure.ADC_HighLevel         = 0;
    ADC1_InitStructure.ADC_VRefSource        = ADC_VREF_SOURCE_INTERNAL;  //ADC1_CFG_M_REF//< Selects Internal Voltage Reference
    ADC1_InitStructure.ADC_IntVRefSource     = ADC_INT_VREF_SOURCE_EXACT; // ADC1_OP// Selects exact Internal Voltage Reference (from Temperature Sensor) 
    ADC1_InitStructure.ADC_Prescaler         = ADC_CLK_div_2;
    ADC1_InitStructure.ADC_DelayGo           = 3;
    ADC1_Init(&ADC1_InitStructure);

    ADC1_Cmd(ENABLE); //включение какнала ADC1