platformio/platformio-core #2088: Home: Libraries: Stats

miki105, 2019-01-31:
PIO Core Call Error: "Error: You are not connected to the Internet.\r\nIf you build a project first time, we need Internet connection to install all dependencies and toolchains."

ivankravets, 2019-02-02:
Could you reproduce this issue after IDE restart?

ivankravets, 2019-02-23:
Please reopen if you still need help

closed 2019-02-23 by ivankravets

j0hncc, 2019-05-15:
I have this problem when going through a cntlm proxy at work.  I use this proxy routinely for git, Intellij, brew, etc, mostly using environment variables http_proxy and https_proxy (I'm using a Mac).

I *was* able to install platformio and then **platformio init -b teensy**, for example, and it worked just fine.

But if I do **platformio lib search LED**  for example, I get the error "You are not connected to the Internet

I have reproduced this several times so I'm wondering if the library search function uses some tool that is not picking up my proxy setting.

ivankravets, 2019-05-15:
@j0hncc please create new issue and make a reference to this. I would be glad to debug it and fix.

nullpainter, 2019-10-05:
I have the same setup. It appears that the library search functionality uses the `HTTP_PROXY` and `HTTPS_PROXY` environment variables. If these are pointing to a bogus proxy server (i.e., cntlm referencing a corporate proxy, while connecting at home), the library search won't work.

Solution is to either remove/rename these environment variables, or tweak cntlm configuration at home by setting `NoProxy *`.