platformio/platformio-core #2105: Home: Could not initialize project

thomasonw, 2019-02-06:
PIO Core Call Error: "The next files/directories have been created in /home/al/Documents/PlatformIO/Projects/ESP32-native\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\nError:"

Fresh install, Ubuntu.  Just using defaults.

Wondering what 'ninclude' is?   include folder was created, but not ninclude. And noting nplatformio.ini , nlin, and nError:    No idea where these extra 'n' are coming from

thomasonw, 2019-02-06:
Update:  I received errors like above doing both ESP32 projects, as well as Arduino projects.  HOWEVER --  I then 'loaded' an example project for each target using the Project Examples button on the home page.   Doing so, the needed files were downloaded and I was able to compile the example projects.

More so, once I did the above for a given platform, I was then able to go back and create a new blank project without and errors.

Hope this helps give some insight.

ivankravets, 2019-02-23:
Could you try to remove `~/.platformio` folder and try again? Could you reproduce this issue? maybe, issue with Internet connection?

ivankravets, 2019-03-07:
Please reopen if you still need help

closed 2019-03-07 by ivankravets

thomasonw, 2019-03-07:
Ivan,

 

Sorry for the delay.  We are doing development, and I need to get to a place where the internet is more responsive before I can do the test you asked for.   In short, we are able to proceed using the steps I noted above.  But I will try to again reproduce the problem and see if it gives some insight.

thomasonw, 2019-03-07:
@ivankravets :  Doing the test above did not reproduce the initial problem, I was able to create a new project after erasing and rebuilding ~/.platformio

ivankravets, 2019-03-08:
Happy coding with PlatformIO!