platformio/platformio-core #2109: pio ci always recompiling  everything

mtfurlan, 2019-02-09:
### Configuration

**Operating system**: Debian testing, Linux moranix 4.19.0-1-amd64 # 1 SMP Debian 4.19.12-1 (2018-12-22) x86_64 GNU/Linux

**PlatformIO Version**: PlatformIO, version 3.6.2

### Description of problem
If I run 
```
pio ci examples/mqtt_pub/ -l . -C platformio.ini --build-dir /tmp/piobuild --keep-build-dir
```
`/tmp/piobuild/platformio.ini` is updated every time that is run, and everything is rebuilt every time.

#### Steps to Reproduce
1. `platfomrio.ini`: 
```
[env:d1_mini]
platform = espressif8266
framework = arduino
board = d1_mini
```
2.
```
pio ci examples/mqtt_pub/ -l . -C platformio.ini --build-dir /tmp/piobuild --keep-build-dir
```

### Actual Results

Rebuilds everything

### Expected Results
Cache anything from the last bui.d

### Additional info

Previous discussion: 
https://community.platformio.org/t/compiling-library-examples-without-recompiling-unchanged-files/5685

ivankravets, 2019-02-23:
Thanks! Please re-test with `$ pio upgrade --dev`.

closed 2019-02-23 by ivankravets

mtfurlan, 2019-03-31:
So that did fix the initial issue, it's not rebuilding everything every time now.
Now I'm noticing that if I change options in the `pio ci` command, `platformio.ini` is not updated to reflect those.
So for example, if I pass `--project-option="build_flags = -Dsomething=42"` and then pass `--project-option="build_flags = -Dsomething=43"`, platformio.ini is not updated.

ivankravets, 2019-04-01:
If you update `platformio.ini`, we will recompile the whole project. So, there is no sense in`--keep-build-dir`.

There are a few workarounds:

1. `pio ci --exclude=platformio.ini` - means, we will remove previous
2. `pio ci --project-conf /path/to/custom/platformio.ini`

commit 522b42c2 (2019-02-23): Fix an issue when `platformio ci` recompiles project if ``--keep-build-dir`` option is passed // Resolve #2109
commit 85679848 (2019-02-23): Fix an issue when `platformio ci` recompiles project if ``--keep-build-dir`` option is passed // Resolve #2109