platformio/platformio-core #2115: Cannot pin version for transitive dependency

OttoWinter, 2019-02-11:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: macOS (`Darwin Kernel Version 18.5.0`)

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.6.4

### Description of problem

Cannot pin version of dependency when another dependency uses it (general library dependency graph problem, not specific to platform).

I recently tried the espressif32@1.6.0 package, but that turned out to be too unstable in a couple of places. Now I want to go back to espressif32@1.5.0 but am hitting a problem downgrading a dependency. 

I'm using the [Async MQTT client](https://github.com/marvinroger/async-mqtt-client) library which uses the [`AsyncTCP`](https://github.com/me-no-dev/AsyncTCP) library. AsyncTCP@1.0.3 is only compatible with espressif32@1.6.0, but AsyncTCP@1.0.1 is only compatible with espressif32@1.5.0 - so I need platformio to pin AsyncTCP to @1.0.1 for using espressif32@1.5.0

However, since AsyncMQTTClient uses AsyncTCP, I cannot manually pin the version in lib_deps:
```ini
lib_deps =
    AsyncMqttClient@0.8.2
    AsyncTCP@1.0.1
```
results in:
```
Scanning dependencies...
Dependency Graph
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.0.3
|-- <AsyncTCP> 1.0.1
```
which fails because AsyncTCP is compiled twice (once with 1.0.3 and once with 1.0.1).
AsyncMqttClient [does not](https://github.com/marvinroger/async-mqtt-client/blob/master/library.json#L25-L28) pin the AsyncTCP version, so I would expect putting `AsyncTCP@1.0.1` would also override the AsyncTCP platformio uses for AsyncMqttClient.

#### Steps to Reproduce

1. New platformio project, platformio.ini (nothing in src/):
```ini
[env:nodemcu-32s]
platform = espressif32@1.5.0
board = nodemcu-32s
framework = arduino
lib_deps =
    AsyncMqttClient@0.8.2
    AsyncTCP@1.0.1
```
2. Observe pio run output:
```
Scanning dependencies...
Dependency Graph
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.0.3
|-- <AsyncTCP> 1.0.1
```

AsyncTCP appears twice with two versions.

3. When compiling (`touch src/main.cpp`), it also appears like 1.0.3 is compiled (resulting in compile error)

### Actual Results

see above

### Expected Results

```
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.0.1
|-- <AsyncTCP> 1.0.1
```

(can pin transitive dependencies in `lib_deps`)

### Additional info

OttoWinter, 2019-02-11:
Turns out order matters for these entries:
```
lib_deps =
    AsyncTCP@1.0.1
    AsyncMqttClient@0.8.2
```

results in 

```
Dependency Graph
|-- <AsyncTCP> 1.0.1
|-- <AsyncMqttClient> 0.8.2
|   |-- <AsyncTCP> 1.0.1
```

I'm gonna keep this issue open because order shouldn't matter for these things IMO

ivankravets, 2019-02-23:
> lib_deps =
>    AsyncMqttClient@0.8.2
>    AsyncTCP@1.0.1

Could you swap 2 of these lines? Please `AsyncTCP@1.0.1` as first.

closed 2019-03-05 by OttoWinter