platformio/platformio-core #2116: heltec_wifi_kit_32 - Vext not defined

affordabletechnology, 2019-02-11:
What kind of issue is this?

- [ x] **Arduino IDE**.

------------------------------------------------------------------

### Configuration
Ide ver 1.8.8

**Operating system**:
Win 10
**PlatformIO Version** (`platformio --version`):
Confirming issue #2107 also applies when compiling with Arduino

### Description of problem
C:\Users\paul\Documents\Arduino\libraries\Heltec_ESP32\src\heltec.cpp:97:10: error: 'Vext' was not declared in this scope pinMode(Vext,OUTPUT);

#### Steps to Reproduce
1. Heltec_ESP32 at version 1.0.2
2. Open example app Vextcontrol and compile
3. Shed a tear and look for a work-around :-(

### Actual Results
C:\Users\paul\Documents\Arduino\libraries\Heltec_ESP32\src\heltec.cpp:97:10: error: 'Vext' was not declared in this scope pinMode(Vext,OUTPUT);

### Expected Results
Compiled Ok!

### If problems with PlatformIO Build System:
no - problem is with Heltec library

### Additional info
Suspect issue relates to Heltec ESP32 Wifi Kit does not support 'Vext' control of OLED display

Looking into putting an ifdef conditional build to eliminate offending code but need to check board schematic first.

elennaro, 2019-02-13:
I have the same issue with PlatformIO, version 3.6.4 CLion 2019.1 EAP

rnunley, 2019-02-20:
This can be fixed by adding #define Vext 21 within #if def(esp32) in heltec.h. I've got a pull request with the fix here: https://github.com/HelTecAutomation/Heltec_ESP32/pull/3. Thanks to Goliath86 for the tip found here: https://github.com/HelTecAutomation/Heltec_ESP32/issues/2.