platformio/platformio-core #2130: Home: Could not initialize project

Pererva, 2019-02-13:
PIO Core Call Error: "The next files/directories have been created in D:\\GITHUB\\mbed_os_examples\\code_snippets\\00. Blink\\01. Blink with variables\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\n\nError: Processing nucleo_f411re (platform: ststm32; board: nucleo_f411re; framework: mbed)\r\n\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f411re.html\r\nPLATFORM: ST STM32 > ST Nucleo F411RE\r\nHARDWARE: STM32F411RET6 100MHz 128KB RAM (512KB Flash)\r\nDEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)\r\nmbed build API internal error\r\npath is on drive C:, start on drive D:\r\n [ERROR] Took 4.84 seconds"

ivankravets, 2019-02-23:
Duplicate of https://github.com/platformio/platform-ststm32/issues/203

Could you reproduce with the latest version of dev/platform? Please PIO Home  > Platforms > Updates

closed 2019-02-23 by ivankravets

Pererva, 2019-02-24:
Fixed. Thank you!