platformio/platformio-core #2144: Pass build flags to the specific scope

ivankravets, 2019-02-18:
Allow passing build flags to the specific scope. See https://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-flags

Example,

```ini
[env:myenv]
board = ...
build_flags =
    ; Only for C files
    CFLAGS: -Wno-old-style-declaration
    ; Only for CPP files
    CXXFLAGS: -fexceptions
    LINKFLAGS: -u __cxa_guard_dummy -u ld_include_panic_highint_hdl 
```

positron96, 2021-12-27:
I wanted to chime in and say that that would be a very useful feature. 
Too often I find myself in a need to add extra py script overcome pio's automatic passing of flags to gcc/g++/ld. 
E.g. this: https://github.com/platformio/platformio-core/issues/594

tai, 2023-01-22:
I'd like to upvote this - it's really cumbersome that I need to prepare an external extra_script= whenever I want to specify some linker flags. Embedded development frequently requires custom linker flags (linker script, etc), so it should be easy as build_flags= to specify one.

The most recent one I was caught is that passing build_flags="-g" does not actually include full debug_info in generated ELF file (AVR target fails to include reference to source/symbol info in included debuginfo). I had to use extra_script= to pass "-g" and get `avr-objdump -CDS` working, and "-g" options is just so common that I feel this should be much easier.

Ariakenom, 2023-10-10:
It seems like this is the intended solution for these issues so I'll make a comment here.
https://github.com/platformio/platformio-core/issues/594
https://github.com/platformio/platformio-core/issues/843

Is it not a bug that pio only passes `-Wl` `build_flags` to the linker when there are more flags that are for linking?