platformio/platformio-core #2147: platformio lib install with no arguments does not install project dependencies

mje-nz, 2019-02-21:
### Configuration

**Operating system**:
macOS 10.12.6

**PlatformIO Version** (`platformio --version`):
3.6.4
### Description of problem
[The documentation](https://docs.platformio.org/en/latest/userguide/lib/cmd_install.html#usage) says that running `platformio lib install` from a project directory will install the libraries listed in the project's `lib_deps`. This behaviour is not implemented (see [here](https://github.com/platformio/platformio-core/blob/6575ddab2654e7f12cd02e18748afe34ca728f27/platformio/commands/lib.py#L105-L107)).

#### Steps to Reproduce
1. From the main directory of a project with external dependencies defined in its default target, run `platformio lib install`.

### Actual Results
The library storage directory is printed and nothing is installed.

### Expected Results
Libraries listed in `lib_deps` are installed.

ivankravets, 2019-02-23:
Libraries will be installed automatically when you try to build project with `pio run`.

Nevertheless, I mark this issue as a feature request.

rileyrg, 2019-03-18:
I've just spent ages finding this. Yes, it would be nice if it was honoured. The --init needs to add the -I for the project to the .clang_complete too.

ivankravets, 2019-05-23:
Please re-test with the latest development version of PIO Core => http://docs.platformio.org/en/latest/installation.html#development-version

or , `pio upgrade --dev` from CLI.

closed 2019-05-23 by ivankravets

mje-nz, 2019-06-12:
Works for me, thanks!

commit c2a549b0 (2019-05-23): Install all project dependencies declared via "lib_deps" option using "platformio lib install" command // Resolve #2147