platformio/platformio-core #2159: Home: Could not load boards list

Navis-Raven, 2019-02-23:
PIO Core Call Error: "Error: Unknown development platform 'C:\\.platformio\\platforms\\ststm32'"

I had many problems with framworks updating so I decided to unistall them and reinstall them, but when uninstalling one of them I get this error :

[image]

Navis-Raven, 2019-02-23:
Adding to that I have many critical problems... 
IntelliSense rebuild doesn't stop
The server crashes
And impossible to build (I think as a consequence of IntelliSense crash)

[image]

And all that makes my PC lagging

Navis-Raven, 2019-02-23:
I've tried to open a ESP32/arduino project, -> same issues

[image]

After waiting 30 min the intellisense rebuild finally stops but I wasn't able to install ST STM32 framwork

Look at the error message
[image]

Navis-Raven, 2019-02-23:
I've tried through another way but this is what it gave

`
Processing disco_l475vg_iot01a (platform: ststm32; board: disco_l475vg_iot01a; framework: mbed)
---------------------------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing toolchain-gccarmnoneeabi @ >=1.60301.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
[Error 5] AccÞs refus: 'C:\\.platformio\\packages\\toolchain-gccarmnoneeabi\\lib\\gcc\\arm-none-eabi\\7.2.1\\cc1.exe' 
Please manually remove the file `C:\.platformio\packages\toolchain-gccarmnoneeabi\lib\gcc\arm-none-eabi\7.2.1\cc1.exe`
[Error 5] AccÞs refus: 'C:\\.platformio\\packages\\toolchain-gccarmnoneeabi\\lib\\gcc\\arm-none-eabi\\7.2.1\\cc1plus.exe' 
Please manually remove the file `C:\.platformio\packages\toolchain-gccarmnoneeabi\lib\gcc\arm-none-eabi\7.2.1\cc1plus.exe`
[Error 5] AccÞs refus: 'C:\\.platformio\\packages\\toolchain-gccarmnoneeabi\\lib\\gcc\\arm-none-eabi\\7.2.1' 
Please manually remove the file `C:\.platformio\packages\toolchain-gccarmnoneeabi\lib\gcc\arm-none-eabi\7.2.1`
[Error 5] AccÞs refus: 'C:\\.platformio\\packages\\toolchain-gccarmnoneeabi\\lib\\gcc\\arm-none-eabi' 
Please manually remove the file `C:\.platformio\packages\toolchain-gccarmnoneeabi\lib\gcc\arm-none-eabi`
[Error 5] AccÞs refus: 'C:\\.platformio\\packages\\toolchain-gccarmnoneeabi\\lib\\gcc' 
Please manually remove the file `C:\.platformio\packages\toolchain-gccarmnoneeabi\lib\gcc`
[Error 5] AccÞs refus: 'C:\\.platformio\\packages\\toolchain-gccarmnoneeabi\\lib' 
Please manually remove the file `C:\.platformio\packages\toolchain-gccarmnoneeabi\lib`
[Error 5] AccÞs refus: 'C:\\.platformio\\packages\\toolchain-gccarmnoneeabi' 
Please manually remove the file `C:\.platformio\packages\toolchain-gccarmnoneeabi`
Error: Could not install 'toolchain-gccarmnoneeabi' with version requirements '>=1.60301.0' for your system 'windows_x86'.

Please try this solution -> http://bit.ly/faq-package-manager
Le processus du terminal s'est achevé avec le code de sortie 1

Le terminal sera réutilisé par les tâches, appuyez sur une touche pour le fermer.

`

ivankravets, 2019-02-23:
See https://community.platformio.org/t/help-me-please-connection-to-server-got-closed/6551/7?u=ivankravets

closed 2019-02-23 by ivankravets

Navis-Raven, 2019-02-23:
@ivankravets  sorry but the problem is still here

`
Processing disco_l475vg_iot01a (platform: ststm32; board: disco_l475vg_iot01a; framework: mbed)
---------------------------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing toolchain-gccarmnoneeabi @ >=1.60301.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
[Error 5] AccÞs refus: 'C:\\.platformio\\packages\\toolchain-gccarmnoneeabi\\lib\\gcc\\arm-none-eabi\\7.2.1\\cc1.exe' 
Please manually remove the file `C:\.platformio\packages\toolchain-gccarmnoneeabi\lib\gcc\arm-none-eabi\7.2.1\cc1.exe`
[Error 5] AccÞs refus: 'C:\\.platformio\\packages\\toolchain-gccarmnoneeabi\\lib\\gcc\\arm-none-eabi\\7.2.1\\cc1plus.exe' 
Please manually remove the file `C:\.platformio\packages\toolchain-gccarmnoneeabi\lib\gcc\arm-none-eabi\7.2.1\cc1plus.exe`
[Error 5] AccÞs refus: 'C:\\.platformio\\packages\\toolchain-gccarmnoneeabi\\lib\\gcc\\arm-none-eabi\\7.2.1' 
Please manually remove the file `C:\.platformio\packages\toolchain-gccarmnoneeabi\lib\gcc\arm-none-eabi\7.2.1`
[Error 5] AccÞs refus: 'C:\\.platformio\\packages\\toolchain-gccarmnoneeabi\\lib\\gcc\\arm-none-eabi' 
Please manually remove the file `C:\.platformio\packages\toolchain-gccarmnoneeabi\lib\gcc\arm-none-eabi`
[Error 5] AccÞs refus: 'C:\\.platformio\\packages\\toolchain-gccarmnoneeabi\\lib\\gcc' 
Please manually remove the file `C:\.platformio\packages\toolchain-gccarmnoneeabi\lib\gcc`
[Error 5] AccÞs refus: 'C:\\.platformio\\packages\\toolchain-gccarmnoneeabi\\lib' 
Please manually remove the file `C:\.platformio\packages\toolchain-gccarmnoneeabi\lib`
[Error 5] AccÞs refus: 'C:\\.platformio\\packages\\toolchain-gccarmnoneeabi' 
Please manually remove the file `C:\.platformio\packages\toolchain-gccarmnoneeabi`
Error: Could not install 'toolchain-gccarmnoneeabi' with version requirements '>=1.60301.0' for your system 'windows_x86'.

Please try this solution -> http://bit.ly/faq-package-manager
Le processus du terminal s'est achevé avec le code de sortie 1

Le terminal sera réutilisé par les tâches, appuyez sur une touche pour le fermer.
`

ivankravets, 2019-02-23:
Could you provide a TeamViewer session to debug this issue? Please ping me at me@ikravets.com

Navis-Raven, 2019-02-23:
I've just installed TeamViewer, I'll contact you on your mail tomorrow. (please ignore my previous mail)