platformio/platformio-core #2164: Build-system doesn't pick up mbed_lib.json files from libraries.

maxgerhardt, 2019-02-24:
What kind of issue is this?
- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** (`platformio --version`): `version 4.0.0a3`

### Description of problem

Taking a look at https://github.com/maxgerhardt/pio-lorawan-example, you can see that inside `lib/mbed-radio-radio-drv/SX127{2,6}` there are `mbed_lib.json` files defining some default value for configuration macros. See e.g. here: 

https://github.com/maxgerhardt/pio-lorawan-example/blob/061615d01c2b97faff79c6322b44dbe2defd7f64/lib/mbed-lora-radio-drv/SX1272/mbed_lib.json#L1-L17

In the `mbed-cli` build system, the last entry would generate the macro definition `#define MBED_CONF_SX1272_LORA_DRIVER_RADIO_VARIANT=SX1272UNDEFINED`. This is then e.g. used here 

https://github.com/maxgerhardt/pio-lorawan-example/blob/061615d01c2b97faff79c6322b44dbe2defd7f64/lib/mbed-lora-radio-drv/SX1272/SX1272_LoRaRadio.cpp#L1465-L1467

However, PlatformIO doesn't generate this definition and trying to build the code without a special `build_flag = -D ..` definition will result in a compilation failure:

```
arm-none-eabi-g++ -o .pioenvs\nucleo_l152re\lib7b3\mbed-lora-radio-drv\SX1272\SX1272_LoRaRadio.o -c -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g1 -mcpu=cortex-m3 -mthumb -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x80000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x14000 -include mbed_config.h -DPLATFORMIO=40000 -DSTM32L152xE -DPIO_FRAMEWORK_MBED_RTOS_PRESENT -DARM_MATH_CM3 -DCOMPONENT_NSPE=1 -DCOMPONENT_PSA_SRV_EMUL=1 -DCOMPONENT_PSA_SRV_IMPL=1 -DDEVICE_ANALOGIN=1 -DDEVICE_ANALOGOUT=1 -DDEVICE_FLASH=1 -DDEVICE_I2C=1 -DDEVICE_I2CSLAVE=1 -DDEVICE_I2C_ASYNCH=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_LPTICKER=1 -DDEVICE_MPU=1 -DDEVICE_PORTIN=1 -DDEVICE_PORTINOUT=1 -DDEVICE_PORTOUT=1 -DDEVICE_PWMOUT=1 -DDEVICE_RTC=1 -DDEVICE_SERIAL=1 -DDEVICE_SERIAL_ASYNCH=1 -DDEVICE_SERIAL_FC=1 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_USTICKER=1 -DTARGET_CORTEX -DTARGET_CORTEX_M -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DTARGET_FF_MORPHO -DTARGET_LIKE_CORTEX_M3 -DTARGET_LIKE_MBED -DTARGET_M3 -DTARGET_NUCLEO_L152RE -DTARGET_RELEASE -DTARGET_STM -DTARGET_STM32L1 -DTARGET_STM32L152RE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CMSIS_RTOS -D__CORTEX_M3 -D__MBED_CMSIS_RTOS_CM -D__MBED__=1 @"C:\Users\Maxi\Desktop\pio_lorawan\.pioenvs\nucleo_l152re\longcmd-153d7654a5a3910d8d8d79694f806803" lib\mbed-lora-radio-drv\SX1272\SX1272_LoRaRadio.cpp
arm-none-eabi-g++ -o .pioenvs\nucleo_l152re\lib7b3\mbed-lora-radio-drv\SX1276\SX1276_LoRaRadio.o -c -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g1 -mcpu=cortex-m3 -mthumb -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x80000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x14000 -include mbed_config.h -DPLATFORMIO=40000 -DSTM32L152xE -DPIO_FRAMEWORK_MBED_RTOS_PRESENT -DARM_MATH_CM3 -DCOMPONENT_NSPE=1 -DCOMPONENT_PSA_SRV_EMUL=1 -DCOMPONENT_PSA_SRV_IMPL=1 -DDEVICE_ANALOGIN=1 -DDEVICE_ANALOGOUT=1 -DDEVICE_FLASH=1 -DDEVICE_I2C=1 -DDEVICE_I2CSLAVE=1 -DDEVICE_I2C_ASYNCH=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_LPTICKER=1 -DDEVICE_MPU=1 -DDEVICE_PORTIN=1 -DDEVICE_PORTINOUT=1 -DDEVICE_PORTOUT=1 -DDEVICE_PWMOUT=1 -DDEVICE_RTC=1 -DDEVICE_SERIAL=1 -DDEVICE_SERIAL_ASYNCH=1 -DDEVICE_SERIAL_FC=1 -DDEVICE_SLEEP=1 -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_SPI_ASYNCH=1 -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_USTICKER=1 -DTARGET_CORTEX -DTARGET_CORTEX_M -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DTARGET_FF_MORPHO -DTARGET_LIKE_CORTEX_M3 -DTARGET_LIKE_MBED -DTARGET_M3 -DTARGET_NUCLEO_L152RE -DTARGET_RELEASE -DTARGET_STM -DTARGET_STM32L1 -DTARGET_STM32L152RE -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CMSIS_RTOS -D__CORTEX_M3 -D__MBED_CMSIS_RTOS_CM -D__MBED__=1 @"C:\Users\Maxi\Desktop\pio_lorawan\.pioenvs\nucleo_l152re\longcmd-153d7654a5a3910d8d8d79694f806803" lib\mbed-lora-radio-drv\SX1276\SX1276_LoRaRadio.cpp
lib\mbed-lora-radio-drv\SX1272\SX1272_LoRaRadio.cpp: In member function 'void SX1272_LoRaRadio::set_sx1272_variant_type()':
lib\mbed-lora-radio-drv\SX1272\SX1272_LoRaRadio.cpp:1466:25: error: 'MBED_CONF_SX1272_LORA_DRIVER_RADIO_VARIANT' was not declared in this scope
radio_variant = MBED_CONF_SX1272_LORA_DRIVER_RADIO_VARIANT;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
lib\mbed-lora-radio-drv\SX1272\SX1272_LoRaRadio.cpp:1466:25: note: suggested alternative: 'MBED_CONF_APP_LORA_RADIO'
radio_variant = MBED_CONF_SX1272_LORA_DRIVER_RADIO_VARIANT;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
MBED_CONF_APP_LORA_RADIO
lib\mbed-lora-radio-drv\SX1276\SX1276_LoRaRadio.cpp: In member function 'void SX1276_LoRaRadio::set_sx1276_variant_type()':
lib\mbed-lora-radio-drv\SX1276\SX1276_LoRaRadio.cpp:1333:25: error: 'MBED_CONF_SX1276_LORA_DRIVER_RADIO_VARIANT' was not declared in this scope
radio_variant = MBED_CONF_SX1276_LORA_DRIVER_RADIO_VARIANT;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
*** [.pioenvs\nucleo_l152re\lib7b3\mbed-lora-radio-drv\SX1272\SX1272_LoRaRadio.o] Error 1
lib\mbed-lora-radio-drv\SX1276\SX1276_LoRaRadio.cpp:1333:25: note: suggested alternative: 'MBED_CONF_APP_LORA_RADIO'
radio_variant = MBED_CONF_SX1276_LORA_DRIVER_RADIO_VARIANT;
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
MBED_CONF_APP_LORA_RADIO
*** [.pioenvs\nucleo_l152re\lib7b3\mbed-lora-radio-drv\SX1276\SX1276_LoRaRadio.o] Error 1
```

I have to manually fix this error by using this `platformio.ini`:

```
build_flags = 
	-D PIO_FRAMEWORK_MBED_RTOS_PRESENT
	-I src/
	-D MBED_CONF_SX1272_LORA_DRIVER_RADIO_VARIANT=SX1272UNDEFINED
	-D MBED_CONF_SX1276_LORA_DRIVER_RADIO_VARIANT=SX1276UNDEFINED
```

Ergo, PlatformIO should automatically pick up on any `mbed_lib.json` files and create these macros, but it doesn't.

#### Steps to Reproduce

1. clone https://github.com/maxgerhardt/pio-lorawan-example 
2. delete the last two `-D` definitions in the `platformio.ini` 
3. see the build fail

### Actual Results

Build failure.

### Expected Results

Successful build because the macros were created.

ivankravets, 2019-03-22:
Sorry, I don't have mbed CLI. Does it add nested folders to CPPPATH? Could you share somewhere verbose build log from this example https://github.com/maxgerhardt/pio-lorawan-example?

ivankravets, 2019-03-23:
Thanks for the report! Please re-test with `pio upgrade --dev`.

maxgerhardt, 2019-03-23:
The compilation is successfuly now for that project (👍), but I don't think it's the correct solution. 

The commit will cause that `-DMBED_CONF_SX1276_LORA_DRIVER_RADIO_VARIANT=SX1276UNDEFINED` is injected only into the compilation of the library sources (e.g. SX1272_LoRaRadio.cpp). But the mbed-cli would add this macro into the `mbed_config.h` (I'll recheck later but I'm pretty sure). 

```cpp
#define MBED_CONF_FILESYSTEM_PRESENT 1 // set by library:mbed-lora-radio-drv
```
This way it is seen by all sources which include `mbed.h` and not only the library's source. E.g., when `main.cpp` is compiled, it doesn't see that macro.

ivankravets, 2019-03-24:
Please `pio upgrade --dev` 😊

commit 3a77bed0 (2019-03-23): Fix an issue when PlatformIO Build System does not pick up "mbed_lib.json" files from libraries // Resolve #2164
commit e33b0fe2 (2019-03-23): Fix an issue when PlatformIO Build System does not pick up "mbed_lib.json" files from libraries // Resolve #2164
commit c1411898 (2019-03-24): Append configuration settings from "mbed_lib.json" to "mbed_config.h" // Resolve #2164
commit 760499c0 (2019-03-24): Append configuration settings from "mbed_lib.json" to "mbed_config.h" // Resolve #2164
closed 2019-03-24 by ivankravets