platformio/platformio-core #2165: Add user-definable monitor options to platformio.ini

maxgerhardt, 2019-02-25:
What kind of issue is this?
- [X] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** (`platformio --version`): `version 4.0.0a3`

### Description of problem

Users may want to give extra options to the execution of the `miniterm.py` script, e.g. `--raw` to enable seeing color codes for the shell (see e.g. [here](https://community.platformio.org/t/some-bizare-characters-prints-on-serial-monitor-with-esp-idf-hello-world-example/5025/11)). This is currently not possible.

Therefore, I suggest to create a new option `monitor_options` in which users can write additional configuration options for `miniterm.py`.

kazooiebombchu, 2019-03-10:
I'd also like this idea. I really want to be able to enable local echo by default when opening monitor if my projects needs it.

kazooiebombchu, 2019-03-10:
Do you accept PRs for this?

ivankravets, 2019-05-25:
Docs http://docs.platformio.org/en/latest/projectconf/section_env_monitor.html

closed 2019-05-25 by ivankravets

del13r, 2023-10-11:
For newbies, add this to env: section of platformio.ini of your project
monitor_raw = yes

commit a84195bb (2019-05-25): Add user-definable monitor options to platformio.ini // Resolve #2165