platformio/platformio-core #2175: Debugger breakpoints are not activated on new debug session start

me21, 2019-03-01:
### Configuration

**Operating system**: Windows 10 x64 v1607 LTSB Russian

**PlatformIO Version** (`platformio --version`): 3.6.4

### Description of problem
The breakpoints I set before I start debug session are not automatically activated when I start a new debug session.

#### Steps to Reproduce

1. Open some "hello world" example.
2. Set a breakpoint somewhere so that it will be hit during program execution.
3. Debug the project.
4. After the execution stops at the beginning of the `main` function, press "Continue" (F5).

### Actual Results
The breakpoint is not hit.

### Expected Results
The breakpoint should be hit.

### Additional info
Workaround: set the breakpoints **after** you start the debug session and the execution stops at the beginning of the `main` function.

ivankravets, 2019-03-03:
Do you use any IDE or CLI?

me21, 2019-03-03:
I use VS Code + PlatformIO extension.