platformio/platformio-core #2185: Add Catch2 as a unit testing framework

davidbitton, 2019-03-04:
- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.
------------------------------------------------------------------

### Configuration

**Operating system**: macos Mojave

**PlatformIO Version** (`platformio --version`): Home 2.0.0·Core 3.6.4

### Description of problem
Add Catch2 as a unit testing framework

### Additional info

https://github.com/catchorg/Catch2

jcw, 2022-04-22:
See also https://github.com/doctest/doctest/issues/506 which tries to run Doctest on embedded, a much faster & simpler varation of catch2. I've been using it on native for all my tests, it's absolutely brilliant, IMNSHO. FWIW. YMMV,

ivankravets, 2022-04-22:
@jcw could you create a separate issue? Also, please point a link in the new issue to https://github.com/doctest/doctest/issues/506

Thanks!