platformio/platformio-core #2186: International character seems to cause errors

lgLindstrom, 2019-03-05:
**Operating system**:
Windows 10 (SE) 

**PlatformIO Version** 
PlatformIO, version 3.6.4

### Description of problem
Problems handling international characters (Swedish) both in project path and when starting debugger.

#### Steps to Reproduce
a) 
1. create a project in a project library containing international characters in the path.
2. build project => "error"

b)
1. create a project in a library NOT containing international characters in the path.
2. start debugger => error

### Actual Results
a) 
```
Processing esp-wrover-kit (framework: arduino; platform: espressif32; board: esp-wrover-kit)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
LibraryManager: Installing webthing-arduino
git version 2.20.1.windows.1
Cloning into 'c:\Users\Lars-Göran Lindström\Documents\PlatformIO\Projects\thingsExampleESP21\.piolibdeps\_tmp_installing-dk8cpn-package'...
remote: Enumerating objects: 44, done.
remote: Counting objects: 100% (44/44), done.
remote: Compressing objects: 100% (34/34), done.
remote: Total 44 (delta 5), reused 23 (delta 1), pack-reused 0
Unpacking objects: 100% (44/44), done.
LibraryManager: Installing id=64
Error: Traceback (most recent call last):
  File "c:\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 107, in cli
    result = (envname, ep.process())
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 207, in process
    result = self._run()
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 302, in _run
    self.verbose)
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\run.py", line 325, in _autoinstall_libdeps
    ctx.invoke(cmd_lib_install, libraries=[lib], silent=not verbose)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context().obj, *args, **kwargs)
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\lib.py", line 103, in lib_install
    library, silent=silent, interactive=interactive, force=force)
  File "c:\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 352, in install
    force=force)
  File "c:\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 723, in install
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "c:\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 213, in _install_from_piorepo
    if app.get_setting("enable_ssl") else dl_data['url'], requirements)
  File "c:\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 508, in _install_from_url
    dlpath = self.download(url, tmp_dir, sha1)
  File "c:\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 178, in download
    dst_path = join(dest_dir, fname)
  File "c:\.platformio\penv\lib\ntpath.py", line 85, in join
    result_path = result_path + p_path
UnicodeDecodeError: 'ascii' codec can't decode byte 0xf6 in position 13: ordinal not in range(128)

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

The terminal process terminated with exit code: 1
```

b)
```
Processing esp32dev (framework: arduino; platform: espressif32; board: esp32dev)
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 &gt; Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
Library Dependency Finder -&gt; http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
No dependencies
Retrieving maximum program size .pioenvsesp32devirmware.elf
Checking size .pioenvsesp32devirmware.elf
Memory Usage -&gt; http://bit.ly/pio-memory-usage
DATA: [ ] 4.1% (used 13336 bytes from 327680 bytes)
PROGRAM: [== ] 19.4% (used 253856 bytes from 1310720 bytes)
[SUCCESS] Took 3.74 seconds

undefinedError: Traceback (most recent call last):

File "&lt;string&gt;", line 70, in main
File "c:\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
return self.main(*args, **kwargs)
File "c:\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
rv = self.invoke(ctx)
File "c:\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "c:\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
return ctx.invoke(self.callback, **ctx.params)
File "c:\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
return callback(*args, **kwargs)
File "c:\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
return f(get_current_context(), *args, **kwargs)
File "&lt;string&gt;", line 90, in cli
File "&lt;string&gt;", line 446, in spawn
File "&lt;string&gt;", line 336, in spawn
UnicodeDecodeError: 'ascii' codec can't decode byte 0xf6 in position 886: ordinal not in range(128)

============================================================
An unexpected error occurred. Further steps:
* Verify that you have the latest version of PlatformIO using
`pip install -U platformio` command
* Try to find answer in FAQ Troubleshooting section
http://docs.platformio.org/page/faq.html
* Report this problem to support@pioplus.com
============================================================
```

**The content of `platformio.ini`:**
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

upload_port = COM6
monitor_port = COM6
monitor_speed = 115200

debug_tool = esp-prog
debug_port = com12

```

**Source file to reproduce issue:**
```
#include <Arduino.h>
void setup()
{
    Serial.begin(115200);
}

void loop()
{
    Serial.println("Hello world!");
    delay(1000);
}
```

### Additional info
See thread https://community.platformio.org/t/suddenly-i-get-this-error/6741

ivankravets, 2019-03-07:
Please open PIO IDE Terminal and type `pio update`. Restart VSCode. Does it work now?

lgLindstrom, 2019-03-07:
Hi
I did as you asked and updated. To make it work I needed manually remove a pioplus.exe file.

Updating and restarting VSCode did not change anything. Still get this error

```
undefinedError: Traceback (most recent call last):
  File "<string>", line 70, in main
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\.platformio\penv\lib\site-packages\click\decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "<string>", line 90, in cli
  File "<string>", line 448, in spawn
  File "<string>", line 336, in spawn
UnicodeDecodeError: 'ascii' codec can't decode byte 0xf6 in position 886: ordinal not in range(128)

============================================================
An unexpected error occurred. Further steps:
* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command
* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html
* Report this problem to support@pioplus.com
============================================================
```

lgLindstrom, 2019-03-07:
Don't know if this helps but.
I tried to develop some code in Java and run into problems with a Java library I used. 
It turned out that java works like this:
- If you use file/string manipulations functions they defaults to use the character sets that your host machine is using.

If you develop your code on one type of computer set to use a certain character set, all tests works okay.
Running same code on another different computer can case all kind of errors.

In my case, the author of the library had to switch all default functions to functions specifying character sets. 

Can this be same kind of problem ??

ivankravets, 2019-05-30:
Could you reproduce this issue with the latest PIO Core 4.0?

```
pio upgrade --dev
```

closed 2019-06-01 by ivankravets