platformio/platformio-core #2192: "Error: no handler found" when cleaning the project

Navis-Raven, 2019-03-06:
- [X ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Win 7 pro 64bit

**PlatformIO Version** (`platformio --version`):

### Description of problem
clicking on on cleaning project, shows up an error "Error: no handler found"
This action may clean the project or not, but this error shows up. 

#### Steps to Reproduce

1. Click on clean project button at the toolbar below

### Actual Results
The message "Error: no handler found" shows up in the output terminal

### Expected Results
The cleaning should success without any error

**The content of `platformio.ini`:**
```ini
[env:disco_l475vg_iot01a]
platform = ststm32
board = disco_l475vg_iot01a
framework = mbed
debug_tool = stlink
build_flags = -D PIO_FRAMEWORK_MBED_RTOS_PRESENT

```

**Source file to reproduce issue:**
```cpp
#include "mbed.h"
#include "rtos.h"
 
Queue<uint32_t, 5> queue;
 
DigitalOut myled(LED1);
 
void queue_isr() {
    queue.put((uint32_t*)2);
    myled = !myled;
}
 
void queue_thread(void const *args) {
    while (true) {
        queue.put((uint32_t*)1);
        Thread::wait(1000);
    }
}
 
int main (void) {
    Thread thread(queue_thread);
    
    Ticker ticker;
    ticker.attach(queue_isr, 1.0);
    
    while (true) {
        osEvent evt = queue.get();
        if (evt.status != osEventMessage) {
            printf("queue->get() returned %02x status\n\r", evt.status);
        } else {
            printf("queue->get() returned %d\n\r", evt.value.v);
        }
    }
}
```

### Additional info
[image]

Navis-Raven, 2019-03-06:
The errors messages shows up also while compiling

[image]

[image]

ivankravets, 2019-03-06:
See https://community.platformio.org/t/in-vsc-task-window-always-popping-up-error-no-handler-found/6511

closed 2019-03-06 by ivankravets

Navis-Raven, 2019-03-06:
excellent