platformio/platformio-core #2195: Home: Could not import project in Win10

dastarke, 2019-03-07:
Hello, 
i installed ther actual versions of vs-code and platformio on Win10. There was no error during installation but i am not able to import any example. The following error message is shown. 

_**Server error: {"message":"[Error 2] Das System kann die angegebene Datei nicht finden: u'C:\\\\Users\\\\Daniel\\\\Documents\\\\PlatformIO'","args":[2,"Das System kann die angegebene Datei nicht finden"],"type":"WindowsError"}**_

I tried a lot but nothing helped. There is no "PlatformIO" in Documents but in C:\users\Daniel.
Hope someone has an idea. 

Daniel

ivankravets, 2019-03-07:
Do you have this folder `C:\\Users\\Daniel\\Documents`?

dastarke, 2019-03-07:
Hello Ivan,

thanks for your reply. Yes it ist he standard documents folder „C:\Users\Daniel\Documents“
or are the double backslash relevant?

Daniel

dastarke, 2019-03-07:
[image]
[image]

dastarke, 2019-03-07:
sorry touched the wrong button.

ivankravets, 2019-03-07:
Could you reproduce this issue after VSCode restart?

dastarke, 2019-03-07:
Yea there is no difference after restarting VSCode.

ivankravets, 2019-03-07:
It seems that you have a permission issue. Could you create `PlatformIO` folder in `C:\Users\Daniel\Documents`? Restart VSCode?

dastarke, 2019-03-07:
I created the folder and restarted VSCode but trying to Import an example Ends in the same issue:

**_Server error: {"message":"[Error 2] Das System kann die angegebene Datei nicht finden: u'C:\\\\Users\\\\Daniel\\\\Documents\\\\PlatformIO\\\\Projects\\\\190307-124128-arduino-blink'","args":[2,"Das System kann die angegebene Datei nicht finden"],"type":"WindowsError"}_**

ivankravets, 2019-03-07:
Do you have antivirus tools in a system?

dastarke, 2019-03-07:
yes Defender.

ivankravets, 2019-03-07:
Could you try to disable it for a while?

dastarke, 2019-03-07:
thanks with disabling everything it works. I will try to find out what exactly is causing the Problem and let you know.

sorry i thought this test was one fo my first...

ivankravets, 2019-03-07:
It seems that Defender does not like when PlatformIO creates folder in background and removes it.

closed 2019-03-07 by ivankravets