platformio/platformio-core #2197: Errors: sh: 1: xtensa-lx106-elf-g++: not found

mmphego, 2019-03-08:
What kind of issue is this?

- [x] **PlatformIO Core**.

------------------------------------------------------------------

### Configuration

**Operating system**:
$ lsb_release -a
```
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.2 LTS
Release:    18.04
Codename:   bionic
```
$ uname -a
```
Linux 4n0m41y 4.15.0-46-generic #49-Ubuntu SMP Wed Feb 6 09:33:07 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 3.6.5

### Description of problem
Errors when compiling for most boards including 

Error: **sh: 1: xtensa-lx106-elf-g++: not found**

#### Steps to Reproduce

1. platformio run -v

### Actual Results
```shell
Processing d1_mini (framework: arduino; lib_deps: 1092, 1089, ArduinoJson, AsyncTCP, ESPAsyncTCP, NTPClient, PubSubClient, SinricClass, WebSockets, RGB-IR-Led-ESP8266=https://github.com/mmphego/RGB-IR-Led-ESP8266/archive/master.zip, Adafruit_IO_Arduino=https://github.com/adafruit/Adafruit_IO_Arduino/archive/master.zip, ArduinoHttpClient=https://github.com/arduino-libraries/ArduinoHttpClient/archive/master.zip; monitor_speed: 115200; platform: espressif8266; board: d1_mini; upload_port: /dev/ttyUSB*)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Storage: /home/mmphego/GitHub/DStv_WiFi_Remote/.piolibdeps
LibraryManager: Installing id=1092
Adafruit MQTT Library @ 0.20.3 is already installed
LibraryManager: Installing id=1089
IRremoteESP8266 @ 2.5.4 is already installed
LibraryManager: Installing id=64
ArduinoJson @ 5.13.4 is already installed
LibraryManager: Installing id=1826
AsyncTCP @ 1.0.3 is already installed
LibraryManager: Installing id=305
ESPAsyncTCP @ 1.2.0 is already installed
LibraryManager: Installing id=551
NTPClient @ 3.1.0 is already installed
LibraryManager: Installing id=89
PubSubClient @ 2.7 is already installed
LibraryManager: Installing id=5509
SinricClass @ 1.0 is already installed
LibraryManager: Installing id=549
WebSockets @ 2.1.2 is already installed
LibraryManager: Installing RGB-IR-Led-ESP8266
RGB-IR-Led-ESP8266 @ 0.0.0 is already installed
LibraryManager: Installing Adafruit_IO_Arduino
Adafruit_IO_Arduino @ 2.7.21 is already installed
LibraryManager: Installing ArduinoHttpClient
ArduinoHttpClient @ 0.0.0 is already installed
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz 80KB RAM (4MB Flash)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Platform incompatible library /home/mmphego/GitHub/DStv_WiFi_Remote/.piolibdeps/AsyncTCP_ID1826
Platform incompatible library /home/mmphego/.platformio/lib/AsyncTCP_ID1826
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
[... 147 lines cut ...]
gcc version 6.5.0 20181026 (Ubuntu 6.5.0-2ubuntu1~18.04) 
```

From gcc-libraries we can find *liblto_plugin.so*
```
(.venv) mmphego@4n0m41y on 10.8.11.168:/usr/lib/gcc/x86_64-linux-gnu/6  [disk: 89%] 
└─ [2019-03-08 09:07:25] $ >>> ls ../4.8/liblto_plugin.so
../4.8/liblto_plugin.so
(.venv) mmphego@4n0m41y on 10.8.11.168:/usr/lib/gcc/x86_64-linux-gnu/6  [disk: 89%] 
└─ [2019-03-08 09:07:29] $ >>> ls ../4.8/liblto_plugin.so
../4.8/liblto_plugin.so
(.venv) mmphego@4n0m41y on 10.8.11.168:/usr/lib/gcc/x86_64-linux-gnu/6  [disk: 89%] 
└─ [2019-03-08 09:07:32] $ >>> ls -lh liblto_plugin.so
lrwxrwxrwx 1 root root 22 Nov 12 10:50 liblto_plugin.so -> liblto_plugin.so.0.0.0
(.venv) mmphego@4n0m41y on 10.8.11.168:/usr/lib/gcc/x86_64-linux-gnu/6  [disk: 89%] 
└─ [2019-03-08 09:07:35] $ >>> ls ../7/liblto_plugin.so
../7/liblto_plugin.so
```

$ java --version
```
openjdk 10.0.2 2018-07-17
OpenJDK Runtime Environment (build 10.0.2+13-Ubuntu-1ubuntu0.18.04.4)
OpenJDK 64-Bit Server VM (build 10.0.2+13-Ubuntu-1ubuntu0.18.04.4, mixed mode)
```

mmphego, 2019-03-08:
Fixed it by deleting `~/.platfomio` and `~/.arduino`, then reinstalled platformio and arduino

closed 2019-03-08 by mmphego

davidsword, 2022-02-19:
thank you. In Docker I deleted `/config/.esphome/platformio/` and it fixed the error for me.