platformio/platformio-core #2201: Debug by CLion

Nullfati, 2019-03-08:
What kind of issue is this?

- [x] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Description of Question
How I can debug by CLion after import project?

maxgerhardt, 2019-03-11:
I show people how to debug stuff in CLion in this tutorial: https://community.platformio.org/t/debugging-stm32-and-esp32-targets-in-an-ide/4048 

Specifics depend on your used board and debugger though. 

But yeah it would be nice to have PlatformIO auto-setup a target like this in the CLion IDE :)

ivankravets, 2019-03-14:
We depend on https://youtrack.jetbrains.com/issue/CPP-15727

ivankravets, 2019-03-14:
We depend on https://youtrack.jetbrains.com/issue/CPP-15727

@anastasiak2512 could you help here?

Nullfati, 2019-03-16:
@maxgerhardt 
Thank you for the info, it's helpful for me.
Yeah, it would be nice builtin to PlatformIO feature "debug by CLion".
Wait when JetBrains fix a bug.

ristomatti, 2019-07-01:
Great news this is a work in progress. I logged in to JetBrains YouTrack and upvoted the issue. I noticed nobody else had done it before me so @Nullfati maybe you could do so also?

ristomatti, 2019-10-09:
@ivankravets Doesn't look very promising. It seems Ilia Motornyi changed the JetBrains' YouTrack issue from _In progress_ to _Open_ and also changed the "Kanban board state" from _Develop_ to _Not on board_...

ivankravets, 2019-10-11:
We work on that together with CLIon Team and @nexoscp.
When we have some news, we will post here.

Sorry for the delay.

ristomatti, 2019-10-12:
Awesome, thanks for the update! I had hard time believing they'd dismiss the opportunity to support PlatformIO but luckily that was not the case after all.

ivankravets, 2019-11-06:
Hi all,

There is a piece of good news here. We are so happy to present the first MVP of the `PlatformIO plugin for CLion`. Big thanks to @JetBrains CLion Team, especially to @elmot for his amazing work.

This is not the final release. We just want to get some feedback and would be thankful if you help us to build a list with the most important features.  

# Demo

## ESP-32

[image]

## RISC-V

[image]

# How to try it?

1. Install the latest [CLion EAP](https://www.jetbrains.com/clion/nextversion/)
2. Download [PlatformIO for Clion Plugin](https://www.dropbox.com/s/bumodympr6rhey0/clion-platformio-plugin-0.0.2.zip?dl=1)
3. Install the plugin: CLion > Preferences > Plugins > ⚙️ > Install  Plugin from Disk ... > Select ZIP archive

[image]

4. Generate PlatformIO project for CLion using docs http://docs.platformio.org/en/latest/ide/clion.html
5. Open a project with CLion and add `PlatformIO` to  configuration using CLion > Run > Edit Configurations... >  [+] > PlatformIO > Give a name `PlatformIO Debug`

[image]

6. Select "PlatformIO Debug" configuration and press "Debug" icon
[image]

7. Happy debugging with PlatformIO and CLion.

# Known issue

1. `[R]` icon (hot-restart) currently only resets + halts target, and does not move execution to initial breakpoint. This is will be fixed in the next version. A temporary solution is to put breakpoints in the code and press "Continue".

sskorol, 2020-01-13:
@ivankravets, could you please share the latest plugin version and PlatformIO Core (dev) reference / setup instruction?

ivankravets, 2020-01-13:
@sskorol see updated instruction above. It is a little bit complicated, this is a 0.0.3 version of PlatformIO CLion plugin. It depends on a special branch of PIO Core which has not been merged yet into the `develop`. We will do this when a public version of Plugin will be available.

If you have any questions, I would be glad to help.

Does it work for you?

sskorol, 2020-01-14:
@ivankravets it works in CLion 2019.3.2 (there's no EAP available right now).
I've successfully tested only "upload" configuration yet, as I have only ESP8266 available right now.

What I'd really love to see in this plugin:
 - pio project type support (create / import);
 - dependency management screen (similar to VSCode);
 - a built-in serial monitor (I don't like an existing CLion plugin);
 - pio ini syntax highlighting and autocompletion;
 - some descriptions for built-in run/debug configurations.

And thanks for your effort!

ivankravets, 2020-01-14:
@sskorol thanks for the feedback! 

> dependency management screen (similar to VSCode);

Do you mean [PIO Home](https://docs.platformio.org/en/latest/home/index.html)?

> built-in serial monitor (I don't like an existing CLion plugin);

Did you try to open a separate terminal window and type `pio device monitor`?

> pio ini syntax highlighting and autocompletion;

1) You need to install INI plugin in CLion
2) There is platformio.ini UI, it has not been released yet. But! If you installed the latest dev-version of PIO Core, it's ready for you. Open CLion terminal and type `pio home`, navigate to PIO Home > Projects

> some descriptions for built-in run/debug configurations.

Could you provide more details?

sskorol, 2020-01-14:
Yes, I meant PIO Home. Just tried `pio home`. See it in a browser now.
Device monitor works via CLI as well. But a general improvement for these 2 features would be adding some icons or tabs into IDE, similar to VSCode:
[image]

This panel is quite convenient. So it would be nice to minimize CLI interactions by creating some UI components at the bottom-left of the screen of CLion:
[image]

In terms of .ini highlighting: I have a corresponding plugin installed. But it still seems like a plain text style:
[image]

Moreover, autocompletion for all the reserved PIO options would really help. Probably, it's also possible to automatically check which platforms and boards are installed and show some suggestions.

Regarding run/debug configuration:
[image]

When I click any of the existing configuration, I see exactly the same settings to the right. So it seems a bit confusing. I can differentiate them only by name. For me it would make a perfect sense to display the underlying commands + references to the official documentation. Or just a couple of sentences about what does a corresponding configuration actually do.

Btw, the following command mentioned in the above instruction doesn't work:
`pio run -b uno --ide clion`

I used this one:
`pio init -b nodemcuv2 --ide clion`

And it'd be really nice to automate it via "New/Import Project" in CLion.

ivankravets, 2020-01-14:
> Btw, the following command mentioned in the above instruction doesn't work:

Sorry, this is my mistake, fixed!

> And it'd be really nice to automate it via "New/Import Project" in CLion.

Yes, there is a progress on this. We will back soon here with updates.

SJ-Innovation, 2020-03-01:
Can we have a link to the repo under which this is being developed/released for future reference?

ivankravets, 2020-03-02:
Yes, see https://github.com/JetBrains/intellij-plugins/tree/master/platformio

sskorol, 2020-03-06:
@ivankravets btw, I noticed a weird behaviour, when it's required to upload a code via OTA. Run/debug configuration is inactive in IDE. Seems like it becomes available only when a required device is connected directly to the USB port. So it might be useful to check if `upload_protocol = espota` is available in the `paltform.io` config and allow building a project in this case.

ivankravets, 2020-03-07:
@sskorol could you provide a screenshot? It should work. Does it work from CLI?

sskorol, 2020-03-10:
@ivankravets that might be related to multiple PIO core conflicts. I had to uninstall everything. But I'm not able to install it again via pip. Seems like the following release was removed: https://github.com/platformio/platformio-core/archive/release/issue-2201-new-clion.zip
Can you share it again? And I'll double check.

closed 2020-03-14 by ivankravets

ivankravets, 2020-03-14:
Finally, we did this! An official [PlatformIO for CLion](https://plugins.jetbrains.com/plugin/13922-platformio-for-clion) is out!  Thanks, JetBrains Team, for their work!

Please follow this documentation:
- https://docs.platformio.org/en/latest/integration/ide/clion.html

There are some known issues. Nevertheless, they are not blocker and we provided temporary solutions for them.

We would be thankful for any feedback.

blotfi, 2020-05-01:
Hi There
Thanks for your work. I installed CLion, VSCode and PIO, 2 days ago.
I am a happy user of IntelliJ IDEs (PyCharm, Android Studio, PHPStorm) and I find that Arduino IDE is so bad... that I was writing my programs on EditPlus or Sublime then compile/upload with Arduino IDE...
So the couple CLion-PIO is something great!

I wrote a tutorial in french today: [https://baghli.blogspot.com/2020/05/clion-et-pateformio-au-lieu-darduino-ide.html](url)
to encourage Arduino/ESP community to use CLion-PIO

The main issues I had:
- I Couldn't install the PIO utilities from CLion alone. I had to first install them with VSCode
- Same for the Platforms and tools (Espressif, AVR...), but I found that if I add to the path the PIO scripts, then CLion can found the PIO utility and install them like VSCode
- Debug is not working (tried on ESP32 for Arduino)
- Syntax highlighting of Arduino, Lib function and names (String, uint8_t, SoftwareSerial, U8g2,...) doesn't work, though the programs compile and upload well. It seems that CLion doesn't found them in its search path. I tried to add them to the CMakeLists, so far without success. For me, it is very important, since I want to clic on a function to go directly to its definition in the c:\Users\xxx\.platformio\lib\yyyy
So if you and CLion team can fix this ASAP, it would be great.

Thanks

elmot, 2020-05-15:
> I Couldn't install the PIO utilities from CLion alone. 

There is a ticket about that https://youtrack.jetbrains.com/issue/CPP-19412

>Same for the Platforms and tools (Espressif, AVR...), but I found that if I add to the path the PIO scripts, then CLion can found the PIO utility and install them like VSCode

There is a ticket about custom PIO location 
https://youtrack.jetbrains.com/issue/CPP-18368

> Debug is not working (tried on ESP32 for Arduino)

Is it physically possible on that platform? 

> Syntax highlighting of Arduino, Lib function and names (String, uint8_t, SoftwareSerial, U8g2,...) doesn't work, though the programs compile and upload well. It seems that CLion doesn't found them [...]

Sounds strange. What is your main file name? Should be something.cpp to express that the code is C++ but not pure C or .ino sketch. Yo may also watch my demo https://www.youtube.com/watch?v=Kz5NQiji9YU&t=5025s

blotfi, 2020-05-15:
Hi elmot
for syntax highlinting, I solved the problem
see : https://youtrack.jetbrains.com/issue/CPP-20192
it seems that we must not use MSVC but MinGW, otherwise CLion will not parse correctly the libraries

commit a22ed402 (2020-03-14): Added initial support for an official "PlatformIO for CLion IDE" plugin // Resolve #2201