platformio/platformio-core #2202: Home: Could not load recent projects

ShmaltS, 2019-03-08:
'utf8' codec can't decode byte 0xd3 in position 9: invalid continuation byte

ShmaltS, 2019-03-08:
[2019-03-08 23:07:54.313] [renderer1] [error] Tree element not found: [object Object]: Error: Tree element not found: [object Object]
    at s.getElementLocation (file:///C:/Users/Учитель/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:89:591)
    at s.isCollapsed (file:///C:/Users/Учитель/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:89:14)
    at $.isCollapsed (file:///C:/Users/Учитель/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:924:572)
    at W.revealMarkersForCurrentActiveEditor (file:///C:/Users/Учитель/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4487:105)
    at W.autoReveal (file:///C:/Users/Учитель/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4486:961)
    at W.onActiveEditorChanged (file:///C:/Users/Учитель/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4484:350)
    at W._register.e (file:///C:/Users/Учитель/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4483:360)
    at d.fire (file:///C:/Users/Учитель/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:73:436)
    at setTimeout (file:///C:/Users/Учитель/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:69:969)

ivankravets, 2019-03-11:
Duplicate of https://github.com/platformio/platformio-core/issues/1959

closed 2019-03-11 by ivankravets