platformio/platformio-core #2206: Home: Could not load recent projects

Schmithtrigger, 2019-03-10:
Server error: {"message":"Invalid `platformio.ini`, project configuration file: 'File contains no section headers.\nfile: C:\\Users\\schmith\\Documents\\PlatformIO\\Projects\\Caudal2polegadas\\platformio.ini, line: 11\n' ;[env:nucleo_f103rb]\\n''","args":["File contains no section headers.\nfile: C:\\Users\\schmith\\Documents\\PlatformIO\\Projects\\Caudal2polegadas\\platformio.ini, line: 11\n' ;[env:nucleo_f103rb]\\n'"],"type":"InvalidProjectConf"}

ivankravets, 2019-03-11:
File contains no section headers.\nfile: C:\Users\schmith\Documents\PlatformIO\Projects\Caudal2polegadas\platformio.ini

closed 2019-03-11 by ivankravets

Schmithtrigger, 2019-03-11:
it work fine when i delete the paste .pioenvs and build, afeter delete de paste .vscode  restart the program vscode. After 5 minutes i have again the paste  .vscode and compile fine and no error.

Thanks.

ivankravets, 2019-03-11:
You have broken `platformio.ini` file in project => https://en.wikipedia.org/wiki/INI_file