platformio/platformio-core #2210: PIO remote issues

dralves, 2019-03-11:
What kind of issue is this?

- [x ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: macos (local) ubuntu 18.04 (docker container) remote

**PlatformIO Version** (`platformio --version`): 3.6.5

### Description of problem

Transcribed from the community forum: https://community.platformio.org/t/pio-remote-issues/6937/2

Hi

I’ve recently changed to pio remote for ci. I’ve encountered a few issues:

Pio 3.6.5 remote fails to use a custom sdkconfig.h. The builder for this framework looks for this file, which isn’t uploaded by default.
Pio 3.6.5 remote fails to build using --force-remote when headers are in include/ . This folder is not automatically uploaded thus making the full remote build fail.
I couldn’t make PIO 4.0.0 (dev version) work with pio remote. Ran into issues with the dreader bytes/string problems. This is not as important since this is a dev version.
By chance I found a work around, which is to kick a build with --force-remote, which will fail but will upload the custom sdkconfig.h and then run a normal remote build afterwards, which will now find that file.

I think PIO remote is super useful, I particularly liked the fact that it doesn’t seem to require opening ports on the firewall and plays nicely with docker/k8s, but it seems to have some sharp edges which makes a bit worried to start using in production ci. Is development ongoing?

Thanks
David

dralves, 2019-03-13:
Hit yet another sharp edge. I'm using a token to authenticate and a long running docker container running just the pio remote agent.

After a couple of days the container lost connection and failed to reconnect with:
```
2019-03-13 09:51:43 [warn] Connection is lost to PIO Remote Cloud. Reconnecting
2019-03-13 09:51:46 [warn] Could not connect to PIO Remote Cloud. Reconnecting...
2019-03-13 09:51:55 [info] Successfully connected
2019-03-13 09:51:55 [info] Authenticating
2019-03-13 09:51:56 [error] Session has expired, please re-login via `pio account login` command
2019-03-13 09:51:56 [info] Successfully disconnected
```

... even though the token is still there and available.

ivankravets, 2019-05-25:
Please re-run 
```
pio upgrade
pio update
```

Everything should work now, including syncing of `include` folder.

Please report if you still have any issues with PIO Remote!

commit 70a0bd72 (2019-05-25): Sync "include" directory for PIO Remote // Resolve #2210
closed 2019-05-25 by ivankravets