platformio/platformio-core #2217: Request removal of library

Duckle29, 2019-03-13:
Hello there. I am working on and maintaining a fork of a library (to update it and add it to platformio) however right after it got registered, I found out you could use spaces in the library name, so I renamed it. This doesn't seem to have been picked up by the update crawler.

The library is ID:6070 and available here:
https://github.com/Hal9k-dk/esp8266-restclient

Preferably if it could be renamed and then updated to 2.0.1, then that'd be the best outcome, but if not, a removal and re-register of it would be just fine :)

Thanks! :)

ivankravets, 2019-03-13:
PlatformIO Library Crawler will automatically pick up your changes. See https://platformio.org/lib/show/6070/ESP8266_REST_client_SSL

It checks manifest every 24h.

closed 2019-03-13 by ivankravets

Duckle29, 2019-03-13:
My brain must be defective. I thought it had been 24h. Thanks :)

pfeerick, 2019-03-13:
Well, it must now have been 24h from it's last check, as it looks like the name up as been updated. ;)