platformio/platformio-core #2222: Home: Could not load platform data

AntumArk, 2019-03-15:
PIO Core Call Error: "Platform C:\\Users\\Caustic_Zinc\\.platformio\\platforms\\ststm32\r\n--------\r\nUpdating ststm32                         @ 2c39824        [a1c78e0]\r\ngit version 2.18.0.windows.1\n\nfatal: not a git repository: '.git'\nError: VCS: Could not process command ['git', 'pull', '--recurse-submodules']"

Error occured when trying to update st stm32 platform.
[image]

ivankravets, 2019-03-21:
How did you install these dev/platforms?

AntumArk, 2019-03-22:
Through platforms window.

ivankravets, 2019-03-22:
Could you uninstall these dev/platforms and install directly from the registry using PIO Home > Platforms > Install.

AntumArk, 2019-04-05:
Installed successfully

ivankravets, 2019-04-05:
Thanks for the confirmation!

closed 2019-04-05 by ivankravets