platformio/platformio-core #2233: supporting Kendryte K210

HelloDB, 2019-03-17:
Consider supporting Kendryte K210?
Relevant information:
https://kendryte.com/
https://forum.kendryte.com/category/5/ide
https://forum.kendryte.com/topic/8/k210-windows-flasher-k-flash-v0-3-0-release

tamusjroyce, 2019-03-18:
Would that also include the https://www.seeedstudio.com/Sipeed-MAix-BiT-for-RISC-V-AI-IoT-p-2872.html prototype kit?

alexwhittemore, 2019-03-19:
I'd imagine the board support package is the easy component of "support," so I suppose so. But I'd definitely be pleased to see some accessible RISC-V hardware!

ivankravets, 2019-03-20:
Yes, we would like to see more hardware based on RISC-V ISA. We've added recently support for GreenWaves GAP8 processor which is based on RISC-V. See https://platformio.org/platforms/riscv_gap

Also, there is support for SiFive boards => https://platformio.org/platforms/riscv

HelloDB, 2019-04-04:
[image]

https://maixduino.sipeed.com/en/

Let the Maix series of development boards (k210 chips) support the Arduino IDE and libraries, making it easy to use a large number of existing open source Arduino libraries for rapid development and prototyping

ivankravets, 2019-04-04:
/CC @btx000, @Neutree

alexwhittemore, 2019-04-04:
@HelloDB Awesome! Thanks for linking that.

Zepan, 2019-04-15:
> /CC @btx000, @Neutree

Hi, I'm zepan from Sipeed. How can we help you to add support for Maix boards?

alexwhittemore, 2019-04-15:
@Zepan Yessss!

I'm no expert on extending PIO, but I think you'll want to:
1) Add a platform. I'm not sure if it makes sense to make a "MAIX" platform or a "Kendryte" platform or what. Or add support to an existing platform. For instance, the HighFive1 is defined in the "RISC-V" platform. https://docs.platformio.org/en/latest/boards/riscv/freedom-e300-hifive1.html (And here's the code: https://github.com/platformio/platform-riscv)
2) Add boards to that platform. Boards at the "platform" level mostly contain info about toolchain definition, not the board support package itself. For instance, at the repo above, you can find the 3 examples in the "boards" directory.
3) Add a "framework" or two to the platform. In the example above, the HighFive1 supports the Freedom E SDK. Presumably, this is where you'd bolt-on the recently released Arduino framework port for the MAIX line, and the "Sipeed_Go" board would support the "Maixduino" framework. 

In reality, since you've already done all the hardest work of porting Arduino to the K210 and built "variants" for the two boards, it shouldn't be TOO much work to plug this all into PlatformIO. But I'm not sure how messy the toolchain support will be for supporting debugging, given that it's work I imagine hasn't been done for the Arduino port.

ivankravets, 2019-04-15:
Hi @Zepan ,

Please take a look at: 

* http://docs.platformio.org/en/latest/platforms/creating_platform.html
* https://github.com/topics/platformio-platform (PlatformIO dev/platform examples)
* https://github.com/platformio/platform-riscv

@alexwhittemore is right. It is better to create a separate dev/platform named as `kendryte`.

btx000, 2019-04-18:
Hi  @ivankravets ,
 I am trying to create a new platform by following the description of this [link](https://docs.platformio.org/en/latest/platforms/creating_platform.html#packages). How can I upload some packages to the https://bintray.com/platformio/dl-packages? Or can I test it locally?

ivankravets, 2019-04-18:
> How can I upload some packages to the https://bintray.com/platformio/dl-packages?

You can share them somewhere and we will deploy to our CDN (this is a temporary issue). Also, you can declare your own `packageRepositories`. See https://docs.platformio.org/en/latest/platforms/creating_platform.html#packages

> Or can I test it locally?

Just place them to `~/.platformio/packages`. Please don't forget to add `package.json` to each package.

btx000, 2019-04-25:
Hi, @ivankravets ,

We have moved the repository to 
https://github.com/sipeed/platform-kendryte210 .

Here is the temporary download link for the toolkits : 
https://bigbits.oss-cn-qingdao.aliyuncs.com/Platform-kendryte210/dl-packages/manifest.json

ivankravets, 2019-04-25:
Hi, @btx000 ,

Great work! 

Actually, you can personally maintain own packages. Just need to add your manifest above https://github.com/sipeed/platform-kendryte210/blob/master/platform.json#L16.

```
    "packageRepositories": [
        "https://bigbits.oss-cn-qingdao.aliyuncs.com/Platform-kendryte210/dl-packages/manifest.json",
        "https://dl.bintray.com/platformio/dl-packages/manifest.json",
        "http://dl.platformio.org/packages/manifest.json"
    ],
```

P.S: Please make SHA1 sums in lowercase in `manifest.json`. I fixed this issue on our side https://github.com/platformio/platformio-core/commit/eab2fd91fd4ae3bca108bc90025187802c4d87b9

ivankravets, 2019-04-25:
> We have moved the repository to

If you will be ready, please make a release and tag in a format `vX.Y.Z`. See https://github.com/platformio/platform-riscv/releases
Also, please provide a list with changes for release on Github. We will show this information later in PIO HOME > Platforms > Kendryte K210 > Changelog

btx000, 2019-04-27:
@ivankravets 
I have released the v0.1.0 version and I think it is now ready to be added to the PIO list. Our download server in China is very slow for users around the world, I hope you can add them to the cdn store. Thank you.

closed 2019-05-03 by ivankravets

ivankravets, 2019-05-03:
@btx000 thanks for the great support and integration!

https://community.platformio.org/t/new-dev-platform-risc-v-kendryte-k210-for-ai-iot/7631

alexwhittemore, 2019-05-03:
I’m so excited, gotta go get me a Maix board now!

Thanks for the swift work :)

ivankravets, 2019-05-03:
If you have any issues, please report to https://github.com/sipeed/platform-kendryte210/issues

commit 41ff1b01 (2019-05-03): Support for Kendryte K210 // Resolve #2233