platformio/platformio-core #2243: Warning "Property !!! WARNING !!! is not allowed" since update from Microsoft C/C++ IntelliSense

wapjoe, 2019-03-22:
What kind of issue is this?
- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.
------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 Pro 64bit, Version: 1803 Build: 17134.648

**PlatformIO Version** (`platformio --version`): Core 3.6.5 @ VSC 1.32.3

### Description of problem
Since the last update of C/C++ IntelliSense from Microsoft, to version 0.22.1, the code is checked in .vscode/c_cpp_properties.json. Since then, a warning for line 2 "Property !!! WARNING !!! is not allowed" and deleting the line only changes it temporarily. The next time VSC is started, the text in line 2 is ""!!!! WARNING !!!": "PLEASE DO NOT MODIFY THIS FILE!  USE https://docs.platformio.org/page/projectconf/section_env_build.html#build-flags"," is entered again and the warning occurs again. When creating a new Platformio project, the line will still be generated.

Is there a setting to prevent this behavior?

#### Steps to Reproduce

1. Open a project
2. Browse to .vscode/c_cpp_properties.json
3. Open it and you getting the warning:
[image]

ivankravets, 2019-03-22:
Thanks! Please re-test with the development version of PlatformIO Core => http://docs.platformio.org/en/latest/ide/vscode.html#platformio-ide-usedevelopmentpiocore

closed 2019-03-22 by ivankravets

wapjoe, 2019-03-22:
@ivankravets Thank you very much for your quick help, the solution works very well. 👍

commit 4ca71e7d (2019-03-22): Project Generator: fixed a warning "Property !!! WARNING !!! is not allowed" for VSCode // Resolve #2243
commit 9c1cc977 (2019-03-22): Project Generator: fixed a warning "Property !!! WARNING !!! is not allowed" for VSCode // Resolve #2243