platformio/platformio-core #2244: Linker error

chessweb01, 2019-03-22:
Building a project for ESP32, I'm getting the following linker error:

*** [.pioenvs\nossl\firmware.elf] Implicit dependency `C:\users\rsel\.platformio\platforms\espressif32\builder\1' not found, needed by target `.pioenvs\nossl\firmware.elf'.

This is my platformio.ini file:

[platformio]
env_default = nossl

[env:ssl]
platform = espressif32
framework = arduino
board = nodemcu-32s
build_flags = -DASYNC_TCP_SSL_ENABLED 0,-DESP32 1
lib_deps = ArduinoJson@~5.13.4

[env:nossl]
platform = espressif32
framework = arduino
board = nodemcu-32s
build_flags = -DESP32 1
lib_deps = ArduinoJson@~5.13.4

ivankravets, 2019-03-22:
> build_flags = -DESP32 1

What does mean `1`? Please remove it from `platformio.ini`.

closed 2019-03-22 by ivankravets

chessweb01, 2019-03-22:
Stupid me, it must be 
 `build_flags = -DESP32=1`
of course