platformio/platformio-core #2253: Home: Could not initialize project

gobyleung, 2019-03-26:
PIO Core Call Error: "The next files/directories have been created in /Development/Arduino/PlatformIOProjects/TestFire\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\nError: Processing m5stack-fire (platform: espressif32; board: m5stack-fire; framework: arduino)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nError: Unknown board ID 'm5stack-fire'\n========================== [ERROR] Took 0.54 seconds =========================="

gobyleung, 2019-03-26:
Try to create a new project for M5Stack Fire but report error unknown board ID

ivankravets, 2019-03-27:
Please update dev/platform via PIO IDE > PIO Home > Platforms > Updates

closed 2019-03-27 by ivankravets