platformio/platformio-core #2256: src_build_flags parsing error

dontsovcmc, 2019-03-27:
[x ] **PlatformIO Core**
------------------------------------------------------------------
### Configuration
**Operating system**: Windows 10

**PlatformIO Version** 3.6.5

### Description of problem
I want to define URL, but I can't

Different variants:
```
src_build_flags = -D TEST_VAR=https://platformio.org  // result: https:
src_build_flags = -D TEST_VAR=https:\/\/platformio.org // https:
src_build_flags = -D TEST_VAR="https://platformio.org" // https:
src_build_flags = -D TEST_VAR=https:platformio.org // https:platformio.org
src_build_flags = -D TEST_VAR=https:/platformio.org // https:/platformio.org
src_build_flags = -D TEST_VAR="https:\/\/platformio.org" //warning: unknown escape sequence: '\/'

```

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino

src_build_flags = -D TEST_VAR="https://platformio.org"
```

**Source file to reproduce issue:**
```
#include <Arduino.h>

#define VALUE_TO_STRING(x) #x
#define VALUE(x) VALUE_TO_STRING(x)
#define VAR_NAME_VALUE(var) #var "="  VALUE(var)

void setup() {
    Serial.begin(9600);
    while(!Serial)
      ;
    
    #pragma message(VAR_NAME_VALUE(TEST_VAR))
    String key = VALUE(TEST_VAR);
    Serial.print(key);
}

void loop() {}
```

ivankravets, 2019-03-27:
It seems like an issue with SCons parser. I'll take a look at it later. A temporary solution is to use [Advanced Scripting](http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html). 

Also, maybe custom options will work for you => http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#custom-options-in-platformio-ini

```ini
[env:esp01_1m]
board = esp01_1m

custom_waterius_host = http://192.168.1.2:4000/data
```

Later in extra script, env.Append(CPPDEFINES=("WATERIUS_HOST", "config.get("my_env", "custom_waterius_host")"))

dontsovcmc, 2019-03-27:
Thanks a lot for the hack! 
It's not a major bug for me.

dontsovcmc, 2019-03-31:
It works!

platformio.ini
```
debug_esp_port=Serial
extra_scripts=extra_script.py
```

extra_script.py
```
from platformio import util
Import("env")

config = util.load_project_config()

env.Append(CPPDEFINES=[ "DEBUG_ESP_WIFI", ("DEBUG_ESP_PORT", config.get("env:esp01_1m", "debug_esp_port"))])
```

ivankravets, 2019-04-18:
Sorry, I can't reproduce this issue. See screenshot below.
[image]

Please upgrade to the latest PIO Core and reopen this issue if you still have this problem.

closed 2019-04-18 by ivankravets