platformio/platformio-core #2274: Home: Could not initialize project

bms1050428813, 2019-04-01:
PIO Core Call Error: "The next files/directories have been created in C:\\Users\\ASUS\\Documents\\PlatformIO\\Projects\\car\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\n\nError:"

ivankravets, 2019-04-01:
Could you provide more details on how to reproduce this issue?

MajidChoi, 2019-04-07:
Could not initialize project
PIO Core Call Error: "The next files/directories have been created in D:\\arm\\Platforimio\\Demo\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\n\nError: Processing genericSTM32F103ZE (platform: ststm32; board: genericSTM32F103ZE; framework: libopencm3)\r\n\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103ZE.html\r\nPLATFORM: ST STM32 > STM32F103ZE (64k RAM. 512k Flash)\r\nHARDWARE: STM32F103ZET6 72MHz 64KB RAM (512KB Flash)\r\nDEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)\r\nIOError: [Errno 2] No such file or directory: 'C:\\\\Users\\\\Majid\\\\.platformio\\\\packages\\\\framework-libopencm3\\\\lib\\\\Makefile':\r\nFile \"C:\\users\\majid\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\main.py\", line 173:\r\nenv.SConscript(\"$BUILD_SCRIPT\")\r\nFile \"C:\\Users\\Majid\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 541:\r\nreturn _SConscript(self.fs, *files, **subst_kw)\r\nFile \"C:\\Users\\Majid\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 250:\r\nexec _file_ in call_stack[-1].globals\r\nFile \"C:\\users\\majid\\.platformio\\platforms\\ststm32\\builder\\main.py\", line 90:\r\ntarget_elf = env.BuildProgram()\r\nFile \"C:\\Users\\Majid\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 224:\r\nreturn self.method(*nargs, **kwargs)\r\nFile \"C:\\users\\majid\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 107:\r\nenv.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))\r\nFile \"C:\\Users\\Majid\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 224:\r\nreturn self.method(*nargs, **kwargs)\r\nFile \"C:\\users\\majid\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 319:\r\nSConscript(env.GetFrameworkScript(f), exports=\"env\")\r\nFile \"C:\\Users\\Majid\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 604:\r\nreturn method(*args, **kw)\r\nFile \"C:\\Users\\Majid\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 541:\r\nreturn _SConscript(self.fs, *files, **subst_kw)\r\nFile \"C:\\Users\\Majid\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 250:\r\nexec _file_ in call_stack[-1].globals\r\nFile \"C:\\users\\majid\\.platformio\\platforms\\ststm32\\builder\\frameworks\\libopencm3\\libopencm3.py\", line 181:\r\nget_source_files(root_dir)\r\nFile \"C:\\users\\majid\\.platformio\\platforms\\ststm32\\builder\\frameworks\\libopencm3\\libopencm3.py\", line 100:\r\nmkdata = parse_makefile_data(join(src_dir, \"Makefile\"))\r\nFile \"C:\\users\\majid\\.platformio\\platforms\\ststm32\\builder\\frameworks\\libopencm3\\libopencm3.py\", line 77:\r\nwith open(makefile) as f:\r\n [ERROR] Took 2.67 seconds"

MajidChoi, 2019-04-07:
> Could you provide more details on how to reproduce this issue?
New project ->finish->"Could not initialize project"