platformio/platformio-core #2279: Home: Could not load installed platforms

exothink, 2019-04-02:
I have uninstalled PIO from VS Code.  Uninstalled Vs Code from Win control panel.  Reinstalled VS Code and PIO.  I still get the following.  I won't load platforms or boards.

PIO Core Call Error: "Error: Traceback (most recent call last):\r\n  File \"c:\\users\\jhe\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 120, in main\r\n    cli(None, None, None)\r\n  File \"c:\\users\\jhe\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 700, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"c:\\users\\jhe\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 680, in main\r\n    rv = self.invoke(ctx)\r\n  File \"c:\\users\\jhe\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1027, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\jhe\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1027, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\jhe\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 873, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"c:\\users\\jhe\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 508, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"c:\\users\\jhe\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\platform.py\", line 223, in platform_list\r\n    expose_packages=False

exothink, 2019-04-02:
Uninstalled PIO, VS Code again.  '.platformio' still existed after the uninstalls.  Deleted it.  Reinstalled all and it is now able to load platforms and boards.

ivankravets, 2019-04-02:
Thanks for the solution. We will try to investigate it.

closed 2019-04-02 by ivankravets