platformio/platformio-core #2298: Home: Could not initialize project

fahrulputra40, 2019-04-07:
PIO Core Call Error: "The next files/directories have been created in E:\\teensy\\stm32\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\n\nError: Processing black_f407vg (platform: ststm32; board: black_f407vg; framework: stm32cube)\r\n\r\nPackageManager: Installing toolchain-gccarmnoneeabi @ >=1.40803.0,<1.40805.0\r\nDownloading\r\nDownloading\r\nUnpacking\r\nUnpacking\r\nPackageManager: Installing framework-stm32cube @ ~1.0.0\r\nDownloading\r\nDownloading\r\nUnpacking\r\nUnpacking\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nError: Unknown board ID 'black_f407vg'\r\n [ERROR] Took 201.41 seconds"

ivankravets, 2019-04-17:
Duplicate of https://github.com/platformio/platformio-core/issues/2299

closed 2019-04-17 by ivankravets

ukey-ltd, 2021-09-01:
Hi Fahrulputra40, 

I sent you a message on Fiverr.

Please check it.

Regards