platformio/platformio-core #2313: Add a way to specify Library name and Library ID on one line in lib_deps

MikeMitterer, 2019-04-14:
What kind of issue is this?

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

------------------------------------------------------------------

### Description of problem
There are several "Stepper" Libraries available
I know that I can use the ID to list it in my dependencies 

```ini
lib_deps =
    ArduinoLog@^1.0.3
    ESP Async WebServer@1.2.0
    ; Stepper (874)
    874@^1.1.0
    ESP8266 and ESP32 Oled Driver for SSD1306 display@^4.0.0
    ArduinoJson@^6.10.0
    ; #1260 OneButton
    1260@f78433c209
    VL53L0X@^1.0.0
```
But this is very annoying. I have to add a comment 'Stepper (874)' to remind me what 874 is!

Maybe you could implement something like this:
`Stepper#874@^1.1.0``

Another solution would be to switch to a Java style package naming convention: https://docs.oracle.com/javase/tutorial/java/package/namingpkgs.html

Thanks

SergiuToporjinschi, 2019-04-16:
Same problem here

ivankravets, 2019-05-03:
Please subscribe to this issue https://github.com/platformio/platformio-core/issues/1824

We are going to deprecate installing libraries by ID in the upcoming PlatformIO Core 4.0. To avoid conflicts between libraries with the same names, you can specify an Owner as a prefix group.

For `874@^1.1.0`, it will be `@arduino-libraries/Stepper@^1.1.0`.

Will it help you?

closed 2019-05-03 by ivankravets

MikeMitterer, 2019-05-04:
Yeah, cool. Everything that makes a packagename unique is a solution. Thanks!