platformio/platformio-core #2318: Home: Could not install platform

AllofLife, 2019-04-15:
PIO Core Call Error: "PlatformManager: Installing Sodaq_wdt\r\ngit version 2.21.0.windows.1\nSodaq_wdt @ 7865d7d has been successfully installed!\n\nCloning into 'C:\\Users\\Administrator\\.platformio\\platforms\\_tmp_installing-krdbmj-package'...\nError: Traceback (most recent call last):\r\n  File \"c:\\users\\administrator\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 120, in main\r\n    cli(None, None, None)\r\n  File \"c:\\users\\administrator\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 700, in __call__\r\n    return self.main(*args, **kwargs)\r\n  File \"c:\\users\\administrator\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 680, in main\r\n    rv = self.invoke(ctx)\r\n  File \"c:\\users\\administrator\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1027, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\administrator\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1027, in invoke\r\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n  File \"c:\\users\\administrator\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 873, in invoke\r\n    return ctx.invoke(self.callback, **ctx.params)\r\n  File \"c:\\users\\administrator\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 508, in invoke\r\n    return callback(*args, **kwargs)\r\n  File \"c:\\users\\administrator\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\platform.py\", line 313, in platform_install\r\n    force=force):\r\n  File \"c:\\users\\administrator\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 71, in install\r\n    p = PlatformFactory.newPlatform(platform_dir)\r\n  File \"c:\\users\\administrator\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 250, in newPlatform\r\n    _instance = platform_cls(join(platform_dir, \"platform.json\"))\r\n  File \"c:\\users\\administrator\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\platform.py\", line 460, in __init__\r\n    self._manifest = util.load_json(manifest_path)\r\n  File \"c:\\users\\administrator\\.platformio\\penv\\lib\\site-packages\\platformio\\util.py\", line 185, in load_json\r\n    with open(file_path, \"r\") as f:\r\nIOError: [Errno 2] No such file or directory: 'C:\\\\Users\\\\Administrator\\\\.platformio\\\\platforms\\\\Sodaq_wdt\\\\platform.json'\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n  `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n  https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n  https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"

AllofLife, 2019-04-15:
did anybody can help me to slove it?

ivankravets, 2019-04-15:
Where did you get `Sodaq_wdt`?

AllofLife, 2019-04-16:
Thank you! i solve it by download the headfile from the github and add it to my code file path ,it run ok.
Firstly I search it at the Platfrom Library and download it .
It seam's that my Vscode can't install it or my code didn't include the file correctly.

closed 2019-04-16 by AllofLife