platformio/platformio-core #2321: Home: Could not initialize project

czecht, 2019-04-16:
Server error: {"message":"[Error 2] The system cannot find the file specified: u'C:\\\\Users\\\\ARDUINO\\\\Documents\\\\PlatformIO'","args":[2,"The system cannot find the file specified"],"type":"WindowsError"}

This is the error when I tried to create a new project.

Windows 10, i7 Intel, 16GB RAM

ivankravets, 2019-04-17:
It seems that antivirus tools block PlatformIO.

czecht, 2019-04-17:
I'm running Microsoft's Antivirus, I never seen it block programs!
Are you sure that it is my antivirus?
Thanks.
Tony

ivankravets, 2019-04-17:
Do you have this folder `C:\\Users\\ARDUINO\\Documents\\PlatformIO`?

czecht, 2019-04-17:
No, it did not create that directory!
Here is where my PlatformIO is - using search command:
search-ms:displayname=Search%20Results%20in%20Documents&crumb=location:C%3A%5CUsers%5CARDUINO%5CDocuments\PlatformIO

czecht, 2019-04-17:
Also, the PC is not using DOUBLE SLASH but a SINGLE SLASH.
Thanks!

ivankravets, 2019-04-18:
It seems that PlatformIO can't create this folder. Maybe permission issue or your antivirus tools block it.

czecht, 2019-04-18:
OK, lets say that you are right.
Why don't I have problems with installing other programs?
Why is it that just your program has this problem?
Thanks.
Tony

ivankravets, 2019-04-18:
VSCode runs PlatformIO Core as subprocess which can lead to an issue with antivirus tools. Could you try to create new project and select another location? Untick "Default location".

czecht, 2019-04-19:
OK, I'll try and let you know if I have any problems.
Thanks!

ivankravets, 2019-05-23:
Please reopen if you still need help

closed 2019-08-25 by ivankravets

LeMasoud, 2021-04-23:
In my case; my system was missing Python so i installed it and voila!

blebee, 2021-05-14:
--------------
I  Make  "New Project" then I found this Error.
"Could not initialize project
PIO Core Call Error: "[WinError 2] The system cannot find the file specified"
---------------------------------
Ryzen5 RAM 8 G  in my laptop don't have anti virus and malware
and My location Project  C:\Users\NITRO 5\Documents\PlatformIO\Projects
Help me please.

PlatformIO-SoeN, 2021-06-24:
I use PlatformIO without problems until the update to PlatformIO IDE 2.3.2
Now I have the same problems ...[WinError 2]....
I Try to:
1. Uninstall PlatformIO from VsCode, uninstall VsCode, uninstall Python
Restart my PC, and start new installation depend on "https://platformio.org/"
again the same problems ...[WinError 2]....

2. Uninstall PlatformIO from VsCode, uninstall VsCode, uninstall Python
Restart my PC, deactivate my Kaspersky-Anti-Virus and start new installation depend on "https://platformio.org/"
again the same problems ...[WinError 2]....

3.
I tried a number of suggested solutions to this problem and also tried older versions of VsCode and PlatformIO IDE
again the same problems ...[WinError 2]....

I have no idea how I should get on there.
I do not understand that this problem is not easy to recreate by the developers of platformIO!

I've really liked the PlatformIO IDE so far

TR7Mike, 2021-09-10:
Just installed PlatforIO and it's now showing as Core 5.2.0b1 / Home 3.3.4 and tried to add a library to my existing project.

Error shows:
Libraries: Installed - PIO Core Call Error: "[WinError 2] The system cannot find the file specified"

Can't see anything of reference in the log files at all. Library installed tab just has spinning waiting icons. 

Path to project is "E:\Documents\Arduino\Projects\MyProjectName"

Yukaira, 2021-10-03:
Hiya, Im getting this same error even after completely disabling windows defender

ive tried reinstalling platformIO multiple times, and multiple versions

Everything related to creating a new project throws this error, and stalls

Any assistance would be appreciated, Thanks

Juoper, 2021-11-24:
I got the same problem and i have the folder what can i do

PlatformIO-SoeN, 2021-11-26:
Hi,
in my case the solution was double backslash in the environment variables

paloj, 2021-11-28:
> Hiya, Im getting this same error even after completely disabling windows defender
> 
> ive tried reinstalling platformIO multiple times, and multiple versions
> 
> Everything related to creating a new project throws this error, and stalls
> 
> Any assistance would be appreciated, Thanks

Same here!

Yukaira, 2021-11-29:
> > Hiya, Im getting this same error even after completely disabling windows defender
> > ive tried reinstalling platformIO multiple times, and multiple versions
> > Everything related to creating a new project throws this error, and stalls
> > Any assistance would be appreciated, Thanks
> 
> Same here!

My issue was special characters in my windows file name. I had to make a new windows user with only alphanumerics in the name and installed it there, worked perfectly there

nabeelkirmani, 2021-12-15:
> My issue was special characters in my windows file name. I had to make a new windows user with only alphanumerics in the name and installed it there, worked perfectly there

I don't have any special characters in my windows file name. I still face the issue. Please assist.

jpolo2021, 2022-04-09:
[image]
可能是window与linux之间的路径出错了吧

MnemonicFish, 2022-06-25:
This solution worked for me...   

1. uninstall platformio from extensions in VScode
3. open windows explorer got to  'Users/' folder 
4. on the menu 'View' then tick the checkbox 'Hidden items' 
5. delete '.platformio' folder
6. then install platformio extension in VScode 

Once re-installed, you should see message 'Platfrom(); Rebuiling IntelliSense Index' (in the bottom footer bar). 

Install development Platform (i.e., Espressif 32) and create project as normal.

IPTRAIN, 2023-01-10:
Hi,

as to my observations it is a fundamental error in different situations ...! Unfortunately the Error notification does not name the file which is basic for the trouble.

E.g. I have tried to import an example  sketch from an Arduino istallation which is definitely not corrupted (fully works in the Arduino IDE eco system) ...!

So I think it is an internal file (of PlatformIO) is not been found.

As so many users are facing this error - what is the official statement from PlatformIO? Or  habe I missed the offical solution / healing for some reason ?

The windows defender (in my installation) is not the problem ... ! During all trouble shooting attempts I had swiitched it off - on the other hand (by experiences from the Arduino IDE), in case of security issuses windows 10 professionel would pop up a system message asking for special allowance ...!

Such a brilliant tool ... and such a frustrating error whithout any general solution ?

Iburtum, 2024-06-08:
i have a esp8266mod mini but im stuck aat the loadfing screen

61pete, 2024-12-12:
For me this was caused by Win10 "ransomware protection" on, which caused everything to be made read only.
Switching to off allowed platformio to create its default project directory in my documents.

javierlf, 2025-10-21:
¿El fin de PlatformIO? Falta de atención por Kravets