platformio/platformio-core #2328: Home: Could not load platform data

Yaooooo, 2019-04-17:
PIO Core Call Error: "Platform C:\\Users\\zhyao\\.platformio\\platforms\\riscv_gap\r\n--------\r\nUpdating riscv_gap                       @ 85427ad\n\nError: VCS: Could not receive an output from `['git', 'ls-remote']` command ({'returncode': 128, 'err': \"fatal: unable to access 'https://github.com/pioplus/platform-riscv_gap.git/': OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to github.com:443 \\n\", 'out': ''})"

ivankravets, 2019-04-17:
>  OpenSSL SSL_connect: SSL_ERROR_SYSCALL i

Please check your Git client.

ivankravets, 2019-04-17:
If you have problem with local Git client, please remove `C:\Users\zhyao\.platformio\platforms\riscv_gap` folder and use stable version from CDN registry. PIO IDE > PIO Home > Platforms > Embedded > Install RISC-V GAP.

Please reopen if it does not work.

closed 2019-04-17 by ivankravets