platformio/platformio-core #2335: PIO ignores upload_port = setting

alexwhittemore, 2019-04-17:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**: macOS 10.14.4

**PlatformIO Version** (`platformio --version`):3.6.6

### Description of problem

The Upload command apparently ignores `upload_port = XXXX` in `platformio.ini`, though output suggests it's being respected. In my case, I'm trying to build hardware UART bootloader support for a SAMD21 using the UF2 bootloader. Upload via an FT232 fails with native USB unplugged, but succeeds with USB plugged in, despite setting `upload_port = <FTDI device>`. I'm certain PIO is using the native USB UART because unplugging the FTDI physically results in a successful upload, even though debug output is correct. 

#### Steps to Reproduce

1. Get a Trinket M0 and install a UF2 bootloader modified such that both hardware UART and native USB Serial are enabled.
__ALTERNATELY__ Presumably it would work to reproduce the issue that you simply have an FT232 adapter plugged in, with some way to monitor the TX/RX lines, while also having plugged in a stock USB serial arduino such as any Adafruit SAMD board. Serial upload won't work over the FT232, of course, but you'll be able to see which device PIO uses.
2. Put together a quick arduino sketch in PIO, configured for your board. Verify it uploads normally with only the native USB port in play.
3. Configure `platformio.ini` to use a hard-coded serial port. For instance, on my machine, `/dev/cu.usbmodem143201` is the trinket M0 USB serial, and `/dev/cu.usbserial-A4016VJ0` is my FT232 adapter. So I set `upload_port = /dev/cu.usbserial-A4016VJ0`
4. Upload code to the device. 

### Actual Results
Software will upload. Please see in-line note below.
```
Processing my-env (framework: arduino; platform: atmelsam; board: my-board)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/my-board.html
PLATFORM: Atmel SAM > Adafruit Trinket M0
HARDWARE: SAMD21E18A 48MHz 32KB RAM (256KB Flash)
DEBUG: CURRENT(jlink) EXTERNAL(atmel-ice, blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 13 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pioenvs/my-env/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]   6.8% (used 2216 bytes from 32768 bytes)
PROGRAM: [          ]   3.8% (used 10088 bytes from 262144 bytes)
Configuring upload protocol...
AVAILABLE: atmel-ice, blackmagic, jlink, sam-ba
CURRENT: upload_protocol = sam-ba
Looking for upload port...
Use manually specified: /dev/cu.usbserial-A4016VJ0
Forcing reset using 1200bps open/close on port /dev/cu.usbserial-A4016VJ0
```
__NOTE__: At this point, I have to manually reset my board by double-tapping a reset button. My board doesn't have a high-pass between FT232 DTR and RST, so this is necessary for UART upload. I'm not sure why it's necessary for native USB UART upload, but it appears to be, still.
```
Waiting for the new upload port...
Uploading .pioenvs/my-env/firmware.bin
Erase flash

Done in 0.819 seconds
Write 10088 bytes to flash (158 pages)
[==============================] 100% (158/158 pages)
Done in 0.068 seconds
Verify 10088 bytes of flash
[==============================] 100% (158/158 pages)
Verify successful
Done in 0.668 seconds
========================================================================= [SUCCESS] Took 7.68 seconds =========================================================================

Terminal will be reused by tasks, press any key to close it.
```

### Expected Results
Failure, because the specified port is an FT232 connected to nothing (or a logic analyzer or such). Note in the log above that my disconnected FT232 is apparently being used, yet upload succeeds when it shouldn't be possible, if that's actually the case.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:my-env]
platform = atmelsam
board = my-board
framework = arduino

debug_init_break = tbreak setup

;debug_tool = atmel-ice
;upload_protocol = atmel-ice

debug_tool = jlink
;upload_protocol = jlink
upload_protocol = sam-ba
upload_port = /dev/cu.usbserial-A4016VJ0
```

**Source file to reproduce issue:**
```cpp
Any example you like, e.g. Blinky.
```

### Additional info

ivankravets, 2019-04-18:
Could you try to add the next lines to `platformio.ini`?

```ini
[env:my-env]
...
board_upload.wait_for_upload_port =
board_upload.use_1200bps_touch = 
```

The empty values are OK which means FALSE. However, I've just fixed it in https://github.com/platformio/platform-atmelsam/commit/d5d6411492fdc58b3a83c683e8b2a31055c986b4 and you will be able to use `board_upload.wait_for_upload_port = false` or `board_upload.wait_for_upload_port = 0` later.

alexwhittemore, 2019-04-18:
```ini

alexwhittemore, 2019-04-18:
Oh - this was part of my problem: the serial device happened to be open with a serial monitor I wasn't aware of, and PlatformIO apparently silently fails on "unable to open port" - it just shows `No device found on cu.usbserial-A4016VJ0` same as if the port works fine, but there's no bootloader behind it.

Now that I've got that sorted out, I'm still getting apparently garbage out. From PlatformIO, the FT232 sends:
```
0x8E (framing error)
0x63 (framing error)
0xD6 (framing error)
0x43 (framing error)
0x00 (framing error)
...
```
(I believe the first byte should be # - 0x32, no?)

Trying the same thing with Arduino, 
```
0x8E (framing error)
0x63 (framing error)
0xF7 (framing error)
0x30 (framing error)
0x70 (framing error)
0x70 (framing error)
0x30 (framing error)
0x70 (framing error)
0x30 (framing error)
0x70 (framing error)
0x70 (framing error)
0x2C (framing error)
0x74 (framing error)
0x23 (framing error)
0x00 (framing error)
...
```

Not quite sure what to make of those.

alexwhittemore, 2019-04-19:
The post above doesn't seem to be relevant. Now that I've sorted out all the other variables in my system, it seems that BOSSA et all just send some stuff before the `#` that I don't understand, but that seems to be standard operating procedure. The long and short of it is that with 
```
upload_port = /dev/myport
board_upload.wait_for_upload_port =
```
PIO works as expected. so then the only question is, why is the second part necessary? Is this expected behavior or is there actually a bug here? If there is, I can try to go looking for it.

ivankravets, 2019-05-23:
Have you resolved this issue?