platformio/platformio-core #2341: Error: Detected unknown package 'contrib-pysite'

mgsmyrk, 2019-04-19:
### Configuration

**Operating system**: Ubuntu 18.04

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.6.6

### Description of problem

After a fresh install of PlatformIO, I cannot run `pio home` because it's unable to install `contrib-pysite`

#### Steps to Reproduce

1. Get a fresh install of PlatformIO with `pip uninstall platformio`, `rm -rf ~/.platformio`, 'pip install -U platformio`.
2. Run `pio home`

### Actual Results

    $ pio home                   
    CorePackageManager: Installing tool-pioplus @ ^2.0.0
    Downloading  [####################################]  100%          
    Unpacking  [####################################]  100%
    tool-pioplus @ 2.1.3 has been successfully installed!
   CorePackageManager: Installing contrib-pysite @ ~2.27.0
    Error: Detected unknown package 'contrib-pysite'

### Expected Results

pio home should install required packages and run successfully.

### Additional info

I've checked that my internet connection is working.  It always fails on this specific package (I can install any other package).

I've also tried downloading a contrib-pysite package from https://bintray.com/platformio/dl-packages/contrib-pysite#files, then installing with `pio platform install file://contrib-pysite-linux_x86_64-2.27.181225.tar.gz`, but this fails with:

    Unpacking  [####################################]  100%
    Error: Could not find one of 'platform.json' manifest files in the package

ivankravets, 2019-04-19:
This is a bug with our CDN and has been just fixed. Please Restart VSCode and re-run `$ pio  update`. Sorry.

closed 2019-04-19 by ivankravets

mgsmyrk, 2019-04-19:
Nice one - that's looking much better now.  Thanks for sorting that out so quickly!

iwaresk, 2019-04-19:
Thank you very much, it's working now.

Jaco1990, 2019-04-21:
Working Thank you

Touma-Kazusa, 2019-08-12:
I face the same problem which shows that "Error: Error: Detected unknown package 'contrib-pysite'".
  
I try 'pio update' but it didn't work it just show me 

Platform Manager
================

Library Manager
===============
Library Storage: /home/rei/.platformio/lib

ivankravets, 2019-08-15:
It means that PlatformIO installer can't download `contrib-pysite` package. Maybe, internet or pythno SSL issue.

You can open PlatormIO IDE Terminal and type:
```
pio home
```

You will see detailed error report.

gs11, 2019-10-09:
The error is a bit confusing and the root cause is never shown due to the exception being caught but not rethrown @ivankravets 
https://github.com/platformio/platformio-core/blob/d2abac9b18ff47cd2ab32312995ba52e78cfd62a/platformio/managers/package.py#L54-L59

Actual exception is:
> EOF occurred in violation of protocol

...which can be remedied by installing package `pyopenssl` in the virtual env.