platformio/platformio-core #2342: Home: Could not initialize project

wernersmook, 2019-04-19:
PIO Core Call Error: "The next files/directories have been created in C:\\Users\\werne\\Documents\\PlatformIO\\Projects\\ESP32_Toets\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\n\nError:"

ivankravets, 2019-04-19:
Could you provide more details? Can you reproduce this issue after VSCode restart?

wernersmook, 2019-04-19:
I have tried several times to create a new ESP32 project, without success.  Each time the process seems to get stuch and there is no internet data flow.  My internet had some problems the past few day, but this can't be the entire problem.  Is is as if PIO just can't download the necessary files

wernersmook, 2019-04-21:
I managed to create a new ESP32 project, thanks.

ivankravets, 2019-04-21:
How did you resolve it?

closed 2019-04-21 by ivankravets

wernersmook, 2019-04-21:
Hi Ivan, I'm not sure what actually helped.  I have uninstalled my original ESP32 platform, restarted VS Code and my internet connection got better last night.