platformio/platformio-core #2346: Home: Libraries: Search

BRAGA96, 2019-04-19:
Server error: {"message":"No module named _dumbwin32proc","args":["No module named _dumbwin32proc"],"type":"ImportError"}

ivankravets, 2019-04-20:
Do you use antivirus tools in a system? What is a version of your Windows OS?

BRAGA96, 2019-04-20:
Only windows deffender, windows 10. But i use vscode insiders.

ivankravets, 2019-04-20:
Could you try to disable it for a while? Maybe, network protection service. Restart VSCode. Does it work now?

BRAGA96, 2019-04-20:
Disconnected, the problem remained. And I can not open the home page.
[Screenshot](data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAABRQAAAN9CAYAAADizfVAAAAgAElEQVR4XuydBZQmxdWG77rO+i4hBAmB8BOSYAkaXAMkSHB3d/dFAyzLYosv7hCCE9yd4MHdYd1d/vPUTg01Pe0zsyPfW+dwqJ2vurvqKemqt2/darPQQgvNMQUREAEREAEREAEREAEREAEREAEREAEREAEREAERyEGgjQTFHJSURAREQAREQAREQAREQAREQAREQAREQAREQAREwBGQoKiGIAIiIAIiIAIiIAIiIAIiIAIiIAIiIAIiIAIikJuABMXcqJRQBERABERABERABERABERABERABERABERABERAgmI92kCf2bPtmuHDbdnp091djurTx+7o3r0ed9SleQhsPXGinTt6tEv6ZseOtvuAATa6bVv372PHjrX9xo938Xu7dbNj+vSxKW3a5Lmt0jRTAl3nzLFBo0bZ3yZPtu/at7ed+/e3Tzt0aLL63m7iRDu7uv01Zp9Pa+dpVdVSx6WWmu9m2m2aTbYWmzHDbhgxwhaYOdPlaZv55rOXO3Vqsv47r8CsOnWq3TJ8uHvcZT162Nm9es2rR1tjvSOT7tsS+25T1s88awh6kAiIgAiIgAiIgAg0MgEJiimAsxb0LXESXaQ9heXnuhc7d7b9+vWzsdXiXXivlaZNs9t/+sn9KSr6+HRZPPPmrbEWS3mf39DlCZ8bbVNJeUpiXLQMLSF9pzlz7KzRo+0fkyY5IXHX/v3tm/btG0WQyNOOt5g0yc4fNco9n/7wUNeujYKxbH9pynGpPs+uz7WNUgGt+KZ5WOfpC3kQNRdBsWx/ylPGuDR/njbNbhs+3NrPmWPn9uplQ3v0KHurwteVeUfmaROtSVBsyvopXKG6QAREQAREQAREQASaKQEJiikVk7UAyTMBb6b1npmtbnPm2JBRo2zDyZNr0s5s08Z2GDCgxrokvEmexWcWz8xMVScos1jKe+8i6RqqPOEzJSjG14C3PG1si9Q87ThME1pbFWk7edKWbV9NOS7V59n1uTYPT6X5mUAe1nn6Qh6mlSoohuVuTEvmuDoo847M0yZak6DYlPWTp98ojQiIgAiIgAiIgAi0BAISFOshKLaECi6bxyVnzLDrhw+3+WbNqnWLq6qq7OzevW3u5rWfQ57FZ1mBpEgZ5uWW58YojwTF+Nr224wbext7nnb82+q+MadNm1rbr4u00zxpG6N95XlufdLkESWS7l+fa+uT50q8Ng/rPH2hvuxa+nidVv6+s2bZdSNG2B+nT6+1zbu+zPJcX2bsyNMmytw3T36bIk1T1k9TlFfPFAEREAEREAEREIHGIJBLUMSH2SaTJxuLeibHbOFhy+EjXbrY9VVV9nX19sMBs2bZxSNHGgsRwp3dutlJgQ+7NaZOtctGjrRus2fbuLZt3XbBFzp3rikXE9pNJ02yv0+eXOc5+Cb8uEMHm1OdOm+eSB61kDioXz9bfMYM23zSJFtw5kwb0a6dXVNVZddVVdnkNm2sy5w5ds7o0S4vccFvNx3NdSk+FPHq96dp02yXCRNstalTrWd1uV/u3Nmu797dXurc2WYHD4jmc9sBAwxPcfuPH29sz5nUpo39q1s3u7hnTxsT2Xb8q5kzbfuJE23DKVPsNzNmuLtSrv917Gj3du1qD3btatML+BI8cPx4O2rs2DrFf79jR9utf3/7sV27mt/CRWEcr50HDHDbVcvw5HnkZe0pU1wd4YeqiPXFxT162I4TJ7q6xuryxU6dbHCvXvZOx441bSntfuGi2lvH4ZMxT/vwfv6KtNXooi6P7624hWAebgf26+fayw4TJhgezULfhNxzq2puiMuEDzp0cO3o1u7dbWRQ//wWFQbO7dnTtp40yY0Z37ZrV8vPZZmBzG8z5tkn9e5tM6rbcpIgkcWEPDB+0Z7oz9EyxOXRWyMuOnOmXTd8uMtDuP06q1wd58yx1adOdePBitOmGVu56aNPMY52727vBW2Se0Xb5RF9+9pGU6a46/vPmmXvduxog3v2tGe7dHHjSNK4FbWOwqMoouiuEybYelOmuHtNa9PGXunUyfmAhUt0rPBj2VaTJtlakWvu6dbN/tO1qxuzvduDOBZZbblMnfEcyg2rtaZOdWM9YzqBd8yHHTrYv7p3twe6dnVjexh8G8c3J22c95q/hm3sjLUTY9w7hPeI1lHYp2gnn7dvb0N79nT+XKMfYYr2saT2VbT84ZiWVE/83fuiTesLefp9n1mzcvtQzDNep1k8RtsQffbtjh0bbbxO6/O9Zs+2q0eMcG0Ly37eHz74sWDnCRNsmenT3dyAHQCftm9vz3fubDdXVbm2Ex2X7u/a1bXlHSZOdGMIb+Gi77S4MXPpnH23jIViffpI0TmUTw9X2jljG+GzDh3s9U6d7JZu3ezNav+dafWTNZbrdxEQAREQAREQAREQgbkEMgXFRWfMcP7CmPTGBRbjB/bta29UT9J+P326XTNihLNsY4J8XJ8+Tlicf9YsGzZihC1VfZ+BvXs7MdILhCtMm2YXjBpV47g9+qzQMqlonqILkKTK9wIov+cRjNIERZYCB48bZ4eMG5fY1q6tqnK+lRAKCdF8Igqx8GehG4aoUItzcYRaFiVxIbpNNKvxM9HmfqtMneqSntG7t207caLLHyHqN66xBEXEgF/PnOmEF4IXJPIKipOqxQAE7DAgGuzdv3/N1u3GFBSLttWGEBTzcIMBoqAXn0OfjLS5K0aMMASRpD5/aN++9mp1nydN2Aa4V7s5c+wX1Yu5sP01tIVLXkExysSXC7+gjF+j2rWrVYa4ctdnezNi9unVfiDj7s1YeU7PnnZNjx41wlPICsEPka8q0pa57ug+feyubt1yCYqMS7uPH2/HjBtXZ1zx+UK4H9SrV81hRuSdjwu7TZiQOHTAhtCQgmKeOuOZeax64cPHLT/W5mnjocCeVPCwjqJ9yl8T1pH/W57nR/tYUh6Klr+xBMWkfp9XUMw7XjemoFh0vM56l8b9njUWcE34ESDr/Ur6Iu+0phQUi/SRrHGHfnVWr17uQzBvqjxzrqyPGmXqU9eIgAiIgAiIgAiIQCUTSBUUWbyeN2qUbTBlipuwcmLuY127Wts5c5zANHDsWLcofbhrVzu8b1+3WEMaw4qFQxT4ja/sB/fr5yyV+KpOiC7uFp4504mNLLII93Xt6kSs4e3a2SIzZ9rJY8bYhOrnc8+ieYouQE7r3dtu7t7dCZ5YYZ0xZozLK//eo39/e7raajJL/EjbIoRF3qCAAdZFb3XqZEtOn25DR46sEWtY5N5QfTJ0NJ8X9Oxpw6qqjIUWVjT4NCSfiB879e/vLJoQbmGHdRABIeDqqiqb2qaN9Zg9232lx7r0lN69a05Czmrwy0+bZjeOGOEsSbHk279fPztg/HhXhwSsmE7o3buWFVOe7XFFeWKtMqxHjzqHwOQVFBGy4P55hw5OqKbdeOvZsM0WFRT9idJZ5SnTf/KIA9RBuDCKXpOHG/2SPvtup061LKfYBjZ01KgaMfnKHj3s/J49nQXcgePG2UHVJ2jTLvbs399+qrZUDBepWN7SPr7o0KGWBS75zmKW1Tajv+cVFG/r3t2G9OzpxpQVp061i0aNqtnOj8Xtk126uFvnacdF88hC94Bx4+zw6o8L/+3UyY7s29e+at/eVp8yxeXFWygljT+INQhMXEu/PmXMGGd1S4jWRdRSMRQnsPS9auRIN474cZY6RLgfPGqUs6gOx0HGcyzVjqm2Vg7fA9wDq+MTxo6tEejzbJtM4he9Nm+dcR3vmye6dLFHu3Rx40WHOXOc9SWncUfZRv3DYvV6Zq9e7h0Du6WnTbNtJ01yh2h4K+OkPIftGY7H9+ljT3fpYlig8d7yYyb5ov4Yy8v2sTRuRcrPffLUU56+kKff5/WhmHe8Lioo+hOls8aeMuN10bGA9DtPnOg+LhAQzdkx8UmHDs7acKGZM22biRPdbgzmSYSQcV5Ged+R4YfaPG2ivhaKefsI4w6W2KeOGeMY8B66qEcPZzG87LRp7l3OBy/ut8uAAc56fs2pU51FKOMSfz+sb1+3C4TAPImxio+TF/bsWabadI0IiIAIiIAIiIAIiEAMgVRBMZygRbcaYn107YgR9rvp0+376kkdk2ACiykWvF5AZHtu1Zw5bqKHCIYQwTU+hBNshI5dBwxwi20f2ErNtlVEFLb5+Elj3jylLUBYnF86cqTbkkxgOzHbCAlZC5CkCThCHNaW60+Z4u7DFlsWp97OcPcJE2xg9UT5uc6dnWA3vm3bOhaKoUVUyJt7+t+iZTuhTx9jIR5vW5bdB6B+7Jgxtle1NZL3mbj+5MnOapEQd8JwnsVnWZ7RXJdZLHEPttKzJZ9A+6Kd0d4aS1As038aQlBMOgAgiz9cEJzo14RoX4y2wdBSNa8vtDx5yG6lP6fIKyiGTKL9M/wtTzsukj/SRrkhICDmEehvJ40Z47YfE8IxLY1VKPojAOIe4bVqi9EkQTEU0hC2otsww3GJBfw/e/Uyxt7wg0V0LPMfkL5s395ZrOYRJZL4pV2bVmdJ90viEH3O5dWiOR9hioa0Ogr7UmilW7aPFc1bmrCcp57y9IU8/T6voBj1j5o0XjeWoFhmvC5aJ6QPmfGBAKHZn1ofd780xhsF7+W877QyY6bPV30FxehuiaQ+En4oRdQPXUtEx8xj+/Rx42aYN+Yoe/frV8eNRJn60jUiIAIiIAIiIAIiIALJBFIFxSQ/enG3i24HjPpT5Jo4v4l8Mca6Aos+Al/l2SbNFr+4UCZPaQsQLFlODyxJwkVNlviRtCjzPtawvCREtwizRfmW4cPdb6E4V2ahhL8wHL/77cjcE3bvd+jgvs4/2rWr8yEVvxm6LuFwIh9aKkVFkdCykrvkWXyW5RnNZVlBET+Utw0fXrPV07fZxhIUy7TVphYUj8QKsdqSLhS7qQN8QQ4aNcpZyxJCK8k8wgLXZLWBooN1mcVxmtCSpx0XzWPY7uKEPH/gDPfNuz08OsaEomhS+eLGiqSy+HEQFxa+z0SFy7hr8whVSc9MuzatzrgfvtOWmD7d+QTlo1PvWbNssZkza23r9oyiH5G43vuvw7/tY126OHE0j8/ZMmNH2T6W1u6KlJ/75KmnPH0hT78vKygmjddl3pN5xp4y43XRsYD0cc9BUORjK5as+FT1VvCkT2O87PTpdvPw4W43ASHPO63MmOnL2dCCYpyPYsoeLVcaZ/8eCsVVn977kX62c2e3kyb8kF2m7nSNCIiACIiACIiACIhAbQKpgmIe3z3+dnH+xUILA9KxzRjfif5ABf4WXShm+bgpk6e0BUjahD1L/EhalGU9L5xEh1t2yiyUkF05sAJR1vsajDZyLBbZ8sxhIlk [... cut ...] d4dbpeXWyrTpF4xKIO205rNvoFt5obuJOZs4SlBrSh2KSqEk+vQVhHEG/RRgBkTLSB47f/+eUUUEx6TmhiJj3ef4paXkP85xHUEwSDePqN7x3WNdZbcFflyZSkiZvuUJhM7QGbdwWr7uLgAiIgAiIgAiIgAiIgAiIgAhUBAEJig0vKCIM/u8js1MvMrvnMbN9tpu7LbRTx/gmxfbnz7+Zu1366jvM1lrZ7NYLzDjkRaH1EQi3OKdZKCaVvLkIinkPTMkjKNbHQjGJU17hLcsiME3gyxL/slpv3Hb3rHtm1X/0mXktI7Pyqt9FQAREQAREQAREQAREQAREQAREoIaABMV4QbGoNVSSpdluW5pdcNJcH4phSLJsWn81s2FnmS04vxppSycQ5/OPMkVFpDgffGkiV5aglNdC0bfxuANSPPs0US7pOdEtznkExbht0f5wFW/ll/d5efIebVtJh7KQLkvgI5/4PY36uwyFvLxtI%E2%80%A6)

closed 2019-08-25 by ivankravets