platformio/platformio-core #2360: Support for MiniCore from MCUDude for ATmega*8 family

procsynth, 2019-04-21:
Hi,

Can you add support for MiniCore from @MCUdude ?
This provide useful parameters for the BOD, clock source/frequency, a custom version of wiring.h implementing support of these parameters in the code and implement some other useful functions for power management.
It also provide options to flash a chip with or without their bootloader of choice : Optiboot.

Supported chips:
- ATmega328
- ATmega168
- ATmega88
- ATmega48
- ATmega8

This core is useful when working on custom hardware.

The repo can be found here : [MCUdude/MiniCore](https://github.com/MCUdude/MiniCore)
and the board manager URL if it helps : [package_MCUdude_MiniCore_index.json](https://mcudude.github.io/MiniCore/package_MCUdude_MiniCore_index.json)

  Thanks in advance.

MCUdude, 2019-04-21:
I'm planning to add support for MiniCore in the future, but first I'd like [platformio/platform-atmelavr #137](https://github.com/platformio/platform-atmelavr/issues/137) to be resolved first.

procsynth, 2019-04-22:
That would be great, your work made mine a great deal easier, thanks !

ivankravets, 2019-04-22:
Duplicate of https://github.com/platformio/platform-atmelavr/issues/137

closed 2019-04-22 by ivankravets